ArduCAM/Arduino #343: SPI Issue: Pi with Mini Multi-Camera Adapter Board

hunterbryant, 2018-07-03:
I recently setup my pi to run a single OV5642 through the standard GPIO pins found in this link: https://github.com/ArduCAM/RaspberryPi/issues/7

I purchased one of the mini adapter boards and am trying to setup the Raspberry Pi to take images from the 4 camera modules. But, I am getting SPI errors for all of the 4 SPI interfaces. I am having trouble figuring out why this is happening. Any help is appreciated

UCTRONICS, 2018-07-03:
@hunterbryant 
Four cameras running simultaneously require external power.

hunterbryant, 2018-07-03:
I have connected power through the micro usb connection on the adapter board

hunterbryant, 2018-07-05:
Any idea? @UCTRONICS

UCTRONICS, 2018-07-06:
@hunterbryant 
How about using two cameras?

ArduCAM, 2018-07-06:
Make sure you can run each camera alone at a time before connecting multiple cameras.

hunterbryant, 2018-07-07:
@UCTRONICS @ArduCAM I can run each camera alone straight through SPI off of the GPIO pins. I can't get one working through the adapter board

hunterbryant, 2018-07-10:
@UCTRONICS @ArduCAM still cannot figure this out. Camera modules will run individually through SPI to the GPIO pins but not through the board

UCTRONICS, 2018-07-22:
@hunterbryant 
Sorry for reply you so late.
Don't worry and  I will test it again tomorrow.