ArduCAM/Arduino #344: Is it Possible to combine Aruducam mini 2mp and small LCD?

2SangH, 2018-07-05:
Hi I am using arducam mini 2mp and ESP8266_Nano_V2. I want get liveview to adafruit 1.44" LCD ( because it have to smaller than shield V2). My questions are here. first, Can I implement Cameara + liveview LCD through Nano V2 in Hardware Capacity? Second, If possible, Can I get continuous frame through arducam 'mini' 2mp? not OV2640? I guess camera's digital pin (D0, D1 ...) is important in continuous view but don't now exactly.. thank you!

2SangH, 2018-07-05:
Thank you for your reply. then If i use ESP12 or ESP32 board (ex, nodeMCU1.0) Can I get continuous frame through arducam mini 2mp? what i am curious is whether use arducam mini 2mp or not 'mini' ...

UCTRONICS, 2018-07-05:
@2SangH 
Yes, The ESP32 has enough pin source. We are working on ArduCAM ESP32 and ArduCAM Shield V2 board.