ArduCAM/Arduino #345: Collect data of arducam mini 2mp image and show image by data recived .

93450101, 2018-07-06:
@ArduCAM  hello lee. It is one step of a big projec , i want collect data of arducam image To be able to display the photo whenever I want.، what should i do @UCTRONICS

UCTRONICS, 2018-07-07:
@93450101 
You can take an image by pressing the button and store it in your SD card.
If you store the image to your SD card, you can see it through your PC anytime.

93450101, 2018-07-07:
@UCTRONICS  hello , thank for your answer , but My country is sanctioned and I can not buy the sd shiled  from amazon's site
On the other hand, the sd card demo takes just 256 images from the image that I think is false, In other words, I want to print all the data in the photo, and I will have them for the next steps in my project.

UCTRONICS, 2018-07-07:
@93450101 
If you don't have an SD card, I advise you use our Host V2 software. You can write the image data to the PC
through serial port. You can refer to our video demo from https://www.youtube.com/watch?v=hybQpjwJ4aA&feature=youtu.be

93450101, 2018-07-13:
@UCTRONICS  thank you for your answer , I want to display the image inside the SD card with the host app. Can you help me?