ArduCAM/Arduino #350: Host application does not display picture

apalmtree, 2018-08-07:
I followed the directions in the "Arducam MINI Camera Demo Tutorial for Arduino" video to set up my 2 MP OV2640 and succeeded in uploading the example to my board. However, I encountered problems when I tried to use the application to capture pictures. When I selected "Open", the application stated "COM is open!". This does not match what I saw in the tutorial (in which the application subsequently displayed more text). Then, I encounter problems when I try to capture a picture. After clicking "Capture", the application states "ArduCAM single", but it does not display the subsequent text or open another window displaying the picture as in the tutorial. The application even displays a size in the lower right side that varies based on where the camera is pointed and apparently changes with light levels. I have double-checked my wiring and cannot find any issues with that. I am using a Sparkfun RedBoard, but as I understand it, it should function like an Arduino Uno.

UCTRONICS, 2018-08-07:
@apalmtree 
Please check your serial port and baud rate

apalmtree, 2018-08-07:
@UCTRONICS Thank you very much for the reply; I will do this. However, it would be helpful if you could explain exactly what you mean by checking my serial port.

apalmtree, 2018-08-08:
@UCTRONICS The baud rate is 921600, as in the tutorial. I am still unsure of how to check the serial port.

UCTRONICS, 2018-08-08:
@apalmtree 
You should choose the right port and connect it 
For example, the UNO board's serial port is COM3  in my computer, so I choose the com3 

[image]

apalmtree, 2018-08-08:
@UCTRONICS The only option I see in the drop-down is COM4, which is what I used to upload the example sketch to my RedBoard.

apalmtree, 2018-08-11:
@UCTRONICS Are there any further tests I might try?

UCTRONICS, 2018-08-13:
@apalmtree 
Do you use the newest Host_V2? Please download the newest from https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app

Please open your device manager to check your serial port.

apalmtree, 2018-08-13:
@UCTRONICS It says it was uploaded 2 months ago, and I downloaded it much more recently than that. Looking at comments on the tutorial on YouTube, it looks like some others have had similar issues with the host application as well.

UCTRONICS, 2018-08-31:
@apalmtree 
Do you use our new code? We update the QT Host V2 and The communication protocol is optimized.
Please download our new code and try it again.