ArduCAM/Arduino #351: Difference between OV2640 Mini 2MP and OV2640 Mini 2MP Plus

zubairzulkifli, 2018-08-09:
Can anyone tell me what is the difference between both module?

UCTRONICS, 2018-08-10:
@zz-zesys 
The 2MP Plus is updated from 2MP. The 2MP has a 384K FIFO while the 2MP Plus has an 8 M FIFO.
And it supports short movie clip and multi-capture.

zz-zesys, 2018-08-10:
@UCTRONICS Thank you so much for the explanation. I'll close this issue then.

closed 2018-08-10 by zz-zesys

jimmywong2003, 2019-03-04:
May I ask further question on the difference between v2640 Mini 2MP (rev B) and v2640 Mini 2MP Plus (rev c)?

Because I tried to use the camera for the demo (https://github.com/NordicPlayground/nrf52-ble-image-transfer-demo).

it doesn't work on the V2640-Mini-2MP-PLUS (work only V2640 Mini 2MP).

When we need to get the fifo length, 

`void arducam_mini_2mp_startSingleCapture()
{
    if(mBytesLeftInCamera == 0)
    {
        while(0)
        {
            arducam_CS_LOW();
            nrf_delay_us(500);
            arducam_CS_HIGH();
            nrf_delay_us(500);   
        }
        arducam_flush_fifo();
        arducam_clear_fifo_flag();
        //Start capture
        arducam_start_capture();

        while(!arducam_get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK));
        
        mBytesLeftInCamera = arducam_read_fifo_length();// - 1;
        nrf_delay_us(500);
        arducam_CS_LOW();
        arducam_set_fifo_burst();
        //arducam_spiWrite(0);
    }
    else
    {
        //nrfSystem.registerError(LS_ERROR, "ARDUCAM: startSingleCapture()", 0, "Last capture not read out!");
    }
}`

`//Read Write FIFO length
//Support ArduCAM Mini only
uint32_t arducam_read_fifo_length(void)
{
  uint32_t len1, len2, len3, length = 0;
  len1 = arducam_read_reg(FIFO_SIZE1);
  len2 = arducam_read_reg(FIFO_SIZE2);
  len3 = arducam_read_reg(FIFO_SIZE3) & 0x7f;
  length = ((len3 << 16) | (len2 << 8) | len1) & 0x07fffff;
  return length;
}`

Is it any special handling if using the 2MP-Plus?

Thanks

miaoshasha, 2024-09-03:
I have the same question @jimmywong2003 , anyone can help?