ArduCAM/Arduino #352: How to configure the registers of OV5642 to modify the image's resolution?

cultivosMushrooms, 2018-08-09:
I need take a picture with OV5642 that has a QVGA resolution, this task for beginning.

I hope you can help me

Thank you so much.

ArduCAM, 2018-08-13:
You have to use our ready to use example to get QVGA image.

uacnix, 2018-08-13:
I have the quite the same problem - It all works nice and easy when using the example, but when I try to change the res to other than QVGA (320x240), it just doesn't do it.
I tried copying the line:
myCAM.OV5642_set_JPEG_size(OV5642_1024x768); delay(1000); 
into many places, but my camera still does pictures only in 320x240:
Code responsible for camera setup and taking the picture. 
read_fifo_burst is basically transferring the image to a server and since it works on the data from the camera, I believe the mistake must be somewhere in camera setup.

```
#include <SPI.h>
#include <Wire.h>
#include <ArduCAM.h>
#include "memorysaver.h"

#define   FRAMES_NUM    0x00
const int CS = 4;
bool is_header = false;
ArduCAM myCAM(OV5642, CS);
uint8_t read_fifo_burst(ArduCAM myCAM);

void setup()
{
	Serial.begin(115200);
	Serial.println(F("INIT"));
	pinMode(CS, OUTPUT);
	SPI.begin();
	myCAM.set_format(JPEG);
	myCAM.InitCAM();
	myCAM.OV5642_set_JPEG_size(OV5642_1024x768); delay(1000);
	myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
	myCAM.clear_fifo_flag();
	myCAM.write_reg(ARDUCHIP_FRAMES, FRAMES_NUM);
	Serial.println(F("takePic"));
	// call the camera to take the shot and write it onto SD card.
	takePic();

	read_fifo_burst(myCAM);
	//Clear the capture done flag
	myCAM.clear_fifo_flag();
	delay(5000);
}

void loop() {
	delay(10000);
}

void takePic() {
	myCAM.flush_fifo();
	myCAM.clear_fifo_flag();
	//Start capture
	myCAM.start_capture();
	while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK));
}
```

UCTRONICS, 2018-08-14:
@uacnix 
Yes, If the resolution cannot be configured normal, you will can't get the right resolution.
Have you tested our video streaming demo?https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_Plus_Video_Streaming/ArduCAM_Mini_5MP_Plus_Video_Streaming.ino

I advise you try the demo we provide and you can change your code following the demo.