ArduCAM/Arduino #353: ACK CMD SPI interface Error! END

Hmamouche, 2018-08-11:
Hi,

I tried to use the ArduCAM library on Arduino Leonardo with Arducam-Mini-2MP (OV2640).
I get an SPI error  when lunching the host application : 
   **ACK CMD SPI interface Error! END**
The pins configuration that I used is exactly like the one showed in:  https://www.youtube.com/watch?v=hybQpjwJ4aA&feature=youtu.be, with the CS at pin 7.
I'm working on mac, but i also tried it on a windows virtual machine, but still the same error.

Do you have any ideas to fix this problem ?

Thanks,

UCTRONICS, 2018-08-13:
@Hmamouche 
I have not used this leonardo board and I advise you test the SPI interface to check if the SPI work fine.

ArduCAM, 2018-08-13:
Can you try the camera on UNO board first before moving to Leonardo

UCTRONICS, 2018-08-15:
@Hmamouche 
If it can work normally on UNO which means the camera module is fine.
Can you give me a timing diagram of the SPI read/write and I will help you check your SPI bus?
You'd better write a demo to test the spi interface independently.

Hmamouche, 2018-08-20:
Hi,

Sorry for the delay.
Yes, the camera module is ok.
Here is the SPI diagrams (MOSI, MISC, SCK) for both boards (UNO and Leonardo), and the problem is in these pins for the Leonardo board,  I didn't find how to define them (number of pins) in the library code.

[image]
[image]

UCTRONICS, 2018-08-21:
@Hmamouche 
I can't find the detail information form your diagram. You can write 0x55 to the 0x00 register and read it in the loop. Such as the below diagram shows.
[image]