ArduCAM/Arduino #354: Pi 3B with Multi-Camera Board V2

myzikimi, 2018-08-16:
Is [Multi-Camera Board V2] and [UC-346] compatible?
The company recommended it and purchased the camera(UC-346).

It works when you connect one, but it does not work when you connect two.
(test by 5V 2A Adapter)

[DIP Swith mode ]
 ON : 2,6  / OFF : 1,3,4,5,7,8

Should this model also remove the resistor?
[UC-365] model is described as a method of removing the resistance on your site.
but, There is no example for [UC-346].

UCTRONICS, 2018-08-16:
@myzikimi 
Yes, it is compatible.
After setting the DIP switch, you should change the python code.
We just control the 7,11,12. You should add control the Pin7 Pin15 and Pin16.
For the detail, please refer to http://www.arducam.com/multi-camera-adapter-module-raspberry-pi/

myzikimi, 2018-08-16:
Thanks for the quick answer.
I already knew about the method and implemented it.

Well.. I am testing with android things.

It is the same source(init: gpios set true => sel cam -  gpio 7: false / gpio 11: false / gpio 12 : true) ,
 but it will fail to recognize the camera if the DIP switch is set to 2 use mode(ON : 2,6 / OFF : 1,3,4,5,7,8).

When DIP switch is set to 1 use mode, one camera works well.

myzikimi, 2018-08-22:
I solved it. It was my bug..

Sorry~ & Thank you~!!