ArduCAM/Arduino #358: ArduCam Mini 2MP with Arduino MKR1000

pranjalranjan299, 2018-09-01:
I would like to connect the ArduCam Mini 2MP with my Arduino MKR1000 for an IOT project. I would like to take pictures from the camera and save it to an SD card connected with the MKR1000 with an SD proto shield or a breakout board. Can you help me with the required pin connections and the code as well? 
I was unable to find much help online at other places so hoping you can help me here!

pranjalranjan299, 2018-09-03:
@UCTRONICS 
That discussion is not fruitful to be honest, you just surrendered saying you've not added support yet. What am I supposed to do? I've bought an ArduCAM and I need to get it working with the MKR1000. Let me know what I can do.

UCTRONICS, 2018-09-03:
@pranjalranjan299 
I have helped you add support it 
Please replace the ArduCAM.h with the file I attached for you.
[ArduCAM.zip](https://github.com/ArduCAM/Arduino/files/2344117/ArduCAM.zip)

pranjalranjan299, 2018-09-03:
@UCTRONICS I have already added the block for the MKR1000 in the ArduCAM.h file. The problem I'm encountering is that the host app v2 crashes after taking a single picture from the camera. Is there any need to change some other code for the MKR?

UCTRONICS, 2018-09-03:
@pranjalranjan299 
Firstly, does your SPI and I2C interface work normally? Have you detected the camera?
I think you should not change any other code. Please let me know more details
You'd better attach some pictures and I will help you check.

UCTRONICS, 2018-09-04:
@pranjalranjan299 
Have you tried the low baud rate? You can try to set the baud rate to 115200 to try.
I have not tested it on MKR1000 platform.

pranjalranjan299, 2018-09-04:
@UCTRONICS yes, tried all baud rates, still stops to respond after a while unfortunately

UCTRONICS, 2018-09-05:
@pranjalranjan299 

OK, what's the system of your computer? It seems that has something to do with the PC system.
We have tested the Host V2 many times and it works normally.

Another way, when it stops, can you still find the serial port of your MKR1000?
You can check it in the device manager. If the serial port disappears, it may cause 
an exception

NipunaSL, 2020-03-01:
Hi Any update with this anyone able to work Arduino MKR1000 and above boards with Arducam Mini 5MP Plus? PLease tell me then how?