ArduCAM/Arduino #359: Adafruit nRF52 feather not compatible with Arducam (OV2640)

MarcelRoehrl, 2018-09-11:
Hey,
the OV2640 cam worked fine on our MEGA, but the requirements for our project changed and we've to work with Adafruit nRF52 feathers. I get the following error when trying to upload a test sketch:

> C:\Users\xxx\Documents\Arduino\libraries\ArduCAM/ArduCAM.h:695:2: error: 'regtype' does not name a type
> 
>   regtype *P_CS;
> 
>   ^
> 
> C:\Users\xxx\Documents\Arduino\libraries\ArduCAM/ArduCAM.h:696:2: error: 'regsize' does not name a type
> 
>   regsize B_CS;
> 
>   ^
> 
> exit status 1
> Error compiling for board Adafruit Bluefruit nRF52 Feather.

I hope I just have to modify the ArduCAM.h file. If so, please give the me neccessary changes.
I hope you can help me fast.
Many thanks in advance

UCTRONICS, 2018-09-12:
@MarcelRoehrl 
Do you use our new ArduCAM.h file? It seems your .h file doesn't support for adafruit feather board.
Please download the file from https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ArduCAM.h

MarcelRoehrl, 2018-09-13:
Hey, thanks for the fast reply. 
I was using the newest library including this file. I replaced it nevertheless but I get the same error.
The file also says that it supports 'Feather M0', nothing about the nRF52.

UCTRONICS, 2018-09-14:
@MarcelRoehrl 
Which board have you chosen? Please attach me some pictures.

MarcelRoehrl, 2018-09-25:
Hey, sorry I was on vacation.
This is the hardware I'm working with, currently just on a breadboard for testing.
Like I said the camera should be fine, because it works on our mega.
We'll propably change to the Arducam Mini 5 MP in the future.
[image]
[image]

MarcelRoehrl, 2018-10-04:
Hey, do you have any updates for me?

UCTRONICS, 2018-10-07:
@MarcelRoehrl 
Hi,
Sorry for my late reply because of the China National Day and we have a holiday.
Please open the Ardu'CAM.h file and add the below code to support the NRF52 board.
#if defined(ARDUINO_ARCH_NRF52)
    #define cbi(reg, bitmask) digitalWrite(bitmask, LOW)
	#define sbi(reg, bitmask) digitalWrite(bitmask, HIGH)
	#define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
	#define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);
	
	#define cport(port, data) port &= data
	#define sport(port, data) port |= data
	
	#define swap(type, i, j) {type t = i; i = j; j = t;}
	
	#define fontbyte(x) cfont.font[x]  
	
	#define regtype volatile uint32_t
	#define regsize uint32_t

#endif

You can download the file from https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ArduCAM.h
I have helped you added it.

MarcelRoehrl, 2018-10-10:
Thanks, I've done that and could compile and upload it to the board.
Now I get a SPI interface error. I was testing the camera with your video streaming example.
I tried different nrf52 boards, different CS pins and connected the VCC pin to 3,3V and USB pin (5V).
I also checked every cable that is connected to the board.
I also connected everything back on a Mega and the camera still works with that board.
I hope you can help me, because I can't get past that SPI interface error.

MarcelRoehrl, 2018-10-16:
I've checked these and they are connected to the right pins. I also tried a different nrf52 board. I temporarily connected a TFT screen that uses SPI and it works. How should I test the SPI interface individually?

UCTRONICS, 2018-10-17:
@MarcelRoehrl 
You can use those demos:
[image]

MarcelRoehrl, 2018-10-17:
I don't have have the hardware for these demos. Like I said, I tested the camera on three different nrf52 and the TFT works on every single one (uses SPI, but not SCL and SDA).
Here is another test code.

```
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#define CAM_CS 11

int bufSize = 150;
byte buf[150];

ArduCAM myCAM( OV2640, CAM_CS );

void setup()
{
  pinMode(CAM_CS, OUTPUT);

  Serial.begin(115200);
  while (!Serial);

  /*
     CAM Initialisierung
  */
  digitalWrite(CAM_CS, LOW);
  delay(1000);
  uint8_t vid, pid;
  uint8_t temp;
  Wire.begin();
  SPI.begin();
  //Reset the CPLD
  myCAM.write_reg(0x07, 0x80);
  delay(100);
  myCAM.write_reg(0x07, 0x00);
  delay(100);
  Serial.println("Arducam initialized");

  while (1)
  {
    //Check if the ArduCAM SPI bus is OK
    myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
    temp = myCAM.read_reg(ARDUCHIP_TEST1);

    if (temp != 0x55) {
      Serial.println("SPI interface Error!");<--------------------------------------------------------I'm getting this error 
      delay(1000); continue;
    } else {
      Serial.println("SPI interface OK"); break;
    }
  }

  while (1)
  {
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))) {
      Serial.println("Can't find OV2640 module!");
      delay(1000); continue;
    }
    else {
      Serial.println("OV2640 detected"); break;
    }
  }

  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  myCAM.OV2640_set_JPEG_size(OV2640_1600x1200);
  digitalWrite(CAM_CS, HIGH);
}
```

I marked where it stops working. I always get the 'SPI interface Error!'.

MarcelRoehrl, 2018-10-30:
Hey, I still didn't get it to work. I've three of these Arducams now, but I get this error everytime. I think it's because of the SDA and SCL port. I'll look into that. Do you have any idea how to fix this?

MarcelRoehrl, 2018-10-31:
I also tried two 4,7 k ohmspullup resistors between the SDA and SCL pins. Also didn't work.

[image]

UCTRONICS, 2018-11-01:
@MarcelRoehrl 
Sorry for my late reply.
Have you tested the I2c interface separately? You can use this i2c demo to test the i2c firstly.
// Wire Master Writer
// by Nicholas Zambetti <http://www.zambetti.com>

// Demonstrates use of the Wire library
// Writes data to an I2C/TWI slave device
// Refer to the "Wire Slave Receiver" example for use with this

// Created 29 March 2006

// This example code is in the public domain.

#include <Wire.h>

void setup()
{
  Wire.begin(); // join i2c bus (address optional for master)
}

byte x = 0;

void loop()
{
  Wire.beginTransmission(4); // transmit to device #4
  Wire.write("x is ");        // sends five bytes
  Wire.write(x);              // sends one byte  
  Wire.endTransmission();    // stop transmitting

  x++;
  delay(500);
}

MarcelRoehrl, 2018-11-06:
Hey, thanks. I tested the code. The writing seems to be working. I added some lines to also print the current  byte value on the serial monitor.
I just tested the cam on an Arduino micro and I get the same SPI interface error there.

MarcelRoehrl, 2018-11-06:
I connected another feather board via SDA and SCL and wrote a receiving part for the code. Now it stops sending after one try. I'll look into that.

```
#include <Wire.h>

void setup() {
  Serial.begin(115200);
  while (!Serial);
  
  Wire.begin(); // join i2c bus (address optional for master)
  Wire.onReceive(receiveEvent);
  Wire.begin(4);
}

void loop() {
}

void receiveEvent(int howMany){
  Serial.println(Wire.read());
}
```

Splitframe, 2019-03-15:
Hi,

I have the same problem. It doesn't seem to be the I2C Bus though. The feather crashes as soon as the 
Arducam library is loaded. I can't even write to serial out of it.

Splitframe, 2019-03-19:
This already does not work:

```
#include <Wire.h>
#include <SPI.h>
#include <bluefruit.h>

uint8_t temp = 0;

int main()
{
  delayMicroseconds(100);
  
  pinMode(11, OUTPUT);
  digitalWrite(11, LOW);
  Serial.begin(9600);
  while (!Serial);
  Serial.println("Init");

  Serial.println("Starting Wire");
  Wire.begin();
  Serial.println("Starting SPI");
  SPI.begin();
  //SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));

  Serial.println("Writing to CAM");

  SPI.transfer(0x07 | 0x80);
  SPI.transfer(0x80);
  delayMicroseconds(100);

  SPI.transfer(0x07 | 0x80);
  SPI.transfer(0x00);
  delayMicroseconds(100);

  while (1)
  {
    SPI.transfer(0x00 | 0x80);
    SPI.transfer(0x55);

    delayMicroseconds(100);

    SPI.transfer(0x00);
    temp = SPI.transfer(0x00);

    if (temp != 0x55) {
      Serial.println("SPI interface Error!");
      delayMicroseconds(1000); 
      continue;
    } else {
      Serial.println("SPI interface OK");
      break;
    }
  }
  SPI.endTransaction();
}

void worker() {
}
```

angeac, 2019-10-19:
@MarcelRoehrl  did you fix the issue? I'm having the same with that error "SPI Interface Error", I think it's something with the wiring but I'm not sure, I already uploaded the code to the feather but I always get the same error