ArduCAM/Arduino #360: OV5642 with SD Card on Arduino UNO

juicejams, 2018-09-12:
I'm using ArduCAM_Mini)5MP Plus Multi Capture2SD.
When I connect the OV5642 M 5MP to the arduino UNO It works fine. When I connect my SD Card it works fine too. However, when I connect both to the SPI Which is PIN 11,12,13 I get SPI interface error. When I comment it out the SPI Interface error both are detected ok but the information I get is too large and causes a over size error. 
I am sharing the PINs 11,12,13 with both OV5642 and the SD card. Is this correct?
How do I switch off the SD Card while reading and turn on the SD Card to save the images ? 
I tried pinmode(SD_CS,OUTPUT); digitalWrite(SD_CS, LOW); but it still doesn't work. 
Please help!

UCTRONICS, 2018-09-13:
@juicejams 
You should notice the SPI CS. The SD card's CS and the camera's CS can't active at the same time.
Before you want to enable the SD card, you should ensure the camera's CS should be HIGH.

UCTRONICS, 2018-09-14:
@juicejams
Do you test our ArduCAM_Mini_5MP_Plus_Multi_Capture2SD.ino without any modification?
Please use our original code to test and ensure the hardware is OK.

juicejams, 2018-09-14:
Yes I did. Without modification I get the SD card error and it keeps looping in that while loop 
 I was told that the SD card memory should be 8mb but now all SD cards are in gb

juicejams, 2018-09-14:
@UCTRONICS  Thank you for your reply. 
When I'm using the sample code, I get SPI Interface error. 
When I add the code pinMode(SD_CS,OUTPUT); digitalWrite(SD_CS,HIGH); I get the error 
ArduCAM Start!
SPI interface OK.
OV5642 detected.
SD Card Error!
SD Card Error!
Even when I add digitalWrite(CS,HIGH); inside the while loop 
//Initialize SD Card
while(!SD.begin(SD_CS))
{
  Serial.println(F("SD Card Error!"));delay(1000);
  digitalWrite(CS, HIGH);
}
I still get the SD Card Error!

`// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use the enhanced functions
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It can  continue shooting and store it into the SD card  in JPEG format
// The demo sketch will do the following tasks
// 1. Set the camera to JPEG output mode.
// 2. Capture a JPEG photo and buffer the image to FIFO
// 3.Write the picture data to the SD card
// 5.close the file
//You can change the FRAMES_NUM count to change the number of the picture.
//IF the FRAMES_NUM is 0X00, take one photos
//IF the FRAMES_NUM is 0X01, take two photos
//IF the FRAMES_NUM is 0X02, take three photos
//IF the FRAMES_NUM is 0X03, take four photos
//IF the FRAMES_NUM is 0X04, take five photos
//IF the FRAMES_NUM is 0X05, take six photos
//IF the FRAMES_NUM is 0X06, take seven photos
//IF the FRAMES_NUM is 0XFF, continue shooting until the FIFO is full
//You can see the picture in the SD card.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
//This demo can only work on OV5640_MINI_5MP_PLUS or OV5642_MINI_5MP_PLUS platform.
#if !(defined (OV5640_MINI_5MP_PLUS)||defined (OV5642_MINI_5MP_PLUS))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define   FRAMES_NUM    0x01
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
const int cardDetect = 8;
#define SD_CS 9
bool is_header = false;
int total_time = 0;
#if defined (OV5640_MINI_5MP_PLUS)
  ArduCAM myCAM(OV5640, CS);
#else
  ArduCAM myCAM(OV5642, CS);
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);
void setup() {
// put your setup code here, to run once:
uint8_t vid, pid;
uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
#else
  Wire.begin();
[... 156 lines cut ...]
  }
  else if ((temp == 0xD8) & (temp_last == 0xFF))
  {
    is_header = true;
    myCAM.CS_HIGH();
    //Create a avi file
    k = k + 1;
    itoa(k, str, 10);
    strcat(str, ".jpg");
    //Open the new file
    outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
    if (! outFile)
    {
      Serial.println(F("File open failed"));
      while (1);
    }
    myCAM.CS_LOW();
    myCAM.set_fifo_burst();   
    buf[i++] = temp_last;
    buf[i++] = temp;   
  }
}
  myCAM.CS_HIGH();
  return 1;
}`

UCTRONICS, 2018-09-14:
I have tested your code and it can work normally.
Please check your SD card. Ensure your SD card has been formatted in FAT32 mode.
The memory of the SD card should not exceed 8M.

juicejams, 2018-09-14:
Which SD Card board are you using to test ? I'm using the following. 
https://learn.sparkfun.com/tutorials/microsd-breakout-with-level-shifter-hookup-guide?_ga=2.60241197.1210830135.1536750459-1825753855.1535094563

UCTRONICS, 2018-09-14:
@juicejams 
 I use our adaptor board 
http://www.uctronics.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/u/4/u4135_4.jpg
http://www.uctronics.com/arducam-mini-multi-camera-adapter-board-for-arduino-raspberry-pi.html

juicejams, 2018-09-14:
@UCTRONICS i managed to get it working halfway. Now my SD Card is writing 1.jpg 2.jpg etc but it is 0 bytes. Previously the SPI cables were long so I changed them to a very short one to solve the SD Card error issue. But now its just writing 0 bytes files. Please help.
Almost there. Thank you for your help !

UCTRONICS, 2018-09-14:
@juicejams 
OK, what the fifo length is? Firstly, you should ensure the camera can capture OK.
You can print the image data to serial port to check if it has the jpeg header "FF D8"
and the end of the image data "FF D9"

juicejams, 2018-09-14:
@UCTRONICS the fifo length is around 30000. my camera can capture ok when using the host_app and the other example. occasionally the D9 FF will not be detected.  I think this might cause some problems but when the file closes it should still have some bytes of data inside right ?
below are the messages I get on serial :
start capture. 
CAM Capture Done.
capture total_time used (in miliseconds):381
The fifo length is :31216
D8 FF
File open
save capture total_time used (in miliseconds):409
start capture.
CAM Capture Done.
capture total_time used (in miliseconds):321
The fifo length is :31648
D9 FF
File close
OK
D8
FF
File open
save capture total_time used (in miliseconds):314

UCTRONICS, 2018-09-14:
@juicejams 
The camera's image seems right. Yes, the SD card should be written some bytes.
Maybe it is still has something to do with your SD slot.

juicejams, 2018-09-14:
I noticed that myCAM.CS_HIGH(); then outFile.write(buf,256); but the SD_CS is never changed to low. won't the SD_CS stay LOW and always read all the signals and get confused ?

juicejams, 2018-09-14:
When I print the HEX bytes of my image. I noticed there are a lot of noise thats why the jpg file is not saving properly and ends up as 0kb. Anyone encounter such a problem ? I tried switching the SD_CS HIGH but no use. 

D8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 FF E0 0 10 4A 46 49 46 0 1 1 1 0 0 0 0 0 0 FF DB 0 3 0 4 3 3 4 3 3 4 4 3 4 5 4 4 5 6 A 7 6 6 6 6 C 9 9 7 A F D F F E D E E 10 12 17 14 10 11 16 11 E E 14 1B 14 16 18 19 1A 1A 1A 10 13 1C 1E 1C 19 1E 17 19 1A 19 BF DB 0 43 1 4 5 5 6 5 6 C 7 7 C 19 11 E 11 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 19 FF C4 0 1F 0 0 1 5 1 1 1 1 1 1 0 0 0 0 0 0 0 0 1 2 3 4 5 6 7 8 9 A B BF C4 0 B5 10 0 2 1 3 3 2 4 3 5 5 4 4 0 0 1 3D 1 2 3 0 4 11 5 12 21 31 41 6 13 11 61 7 22 31 14 32 81 91 A1 8 23 2 B1 81 15 12 D1 F0 24 33 62 72 82 9 A 16 17 18 19 1A 25 26 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 27 28 29 2A 34 35 36 37 38 39 3A 3 44 45 46 47 48 49 4A 53 54 55 56 57 58 59 5A 63 64 65 66 67 68 69 6A 73 74 75 76 77 78 79 7A 83 84 85 86 87 88 89 8A 92 93 94 95 96 97 98 99 9A A2 A3 A4 A5 A6 A7 A8 A9 AA B2 B3 B4 B5 B6 B7 B8 B9 BA C2 C3 C4 C5 C6 C7 C8 C9 CA D2 D3 D4 D5 D6 D7 D8 D9 DA E1 E2 E3 E4 E5 E6 E7 E8 E9 EA F1 F2 F3 F4 F5 F6 F7 F8 F9 FA FF C4 0 1F 1 0 3 1 1 1 1 1 1 1 1 1 0 0 0 0 0 0 1 2 3 4 5 6 7 8 9 A B BF C4 0 B5 11 0 2 1 2 4 4 3 4 7 5 4 4 0 1 2 37 0 1 2 3 11 4 5 21 31 6 12 1 51 7 61 71 13 22 32 81 8 14 42 91 A1 B1 C1 9 23 33 52 F0 15 22 72 D1 A 16 24 34 E1 25 F1 17 18 19 1A 26 27 28 29 2A 35 36 37 38 39 3A 43 44 45 46 47 48 49 4A 53 54 55 56 57 58 59 5A 63 64 65 66 67 68 69 6A 73 74 75 76 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 77 78 79 7A 82 83 84 85 86 87 88 89 8A 92 93 94 95 96 97 98 99 9A A2 A3 A4 A5 A6 A7 A8 A9 AA B2 B3 B4 B5 B6 B7 B8 B9 BA C2 C3 C4 C5 C6 C7 C8 C9 CA D2 D3 D4 D5 D6 D7 D8 D9 DA E2 E3 E4 E5 E6 E7 E8 E9 EA F2 F3 F4 F5 F6 F7 F8 F9 FA FF C0 0 11 8 0 F0 1 40 3 1 21 0 2 11 1 3 11 1 BF DA 0 C 3 1 0 2 11 3 11 0 3F 0 BB 3E 36 DC F5 3 B4 AF 9B 47 B6 42 AB CD 39 B8 AD 8C 1B F7 84 5E 4D 4C 58 22 92 91 8 CB 96 6D CF 59 97 73 65 B0 2B 26 28 99 A 49 31 54 DD A9 22 EE 57 90 A1 6A 8C 92 16 88 89 B2 BC B2 B3 55 FC EA 66 77 2A CD 73 E8 6A B4 B7 65 56 A8 86 C8 85 F9 CD 4E 9A 8F 3D 69 91 CC 5C 87 12 3E B5 A3 E A6 3F BD A C5 73 1A 10 6A C5 71 CD 6C 5A EB C5 7F A B2 94 B 13 36 AD 7C 55 24 7D 24 35 D0 58 F8 AC 36 4 D8 35 CA E0 E1 AA 9 32 CF 73 A2 B6 BE 86 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 E8 7E E9 F9 F4 AB 15 AE 61 B1 4A 71 B4 8E 26 B9 42 8A F4 2E 48 51 4C 2 8A 0 2B 98 F1 1F 81 34 9F 11 C4 7C F8 16 29 BB 48 82 83 A 73 70 67 91 6B 5F 7 F5 3D 3A 46 6B 0 2E A1 B5 15 8A CB E1 BD 2 D1 B1 35 B3 8F 82 BC 89 D2 EC 7A B1 97 36 C3 A3 D3 AE 17 AC 2D B9 55 93 65 22 FF 0 1 AE 4E 53 74 57 94 34 7D 57 15 83 AD 1C 81 C 9E 95 49 1 F4 DE 33 15 A5 AB 59 24 73 5C 12 A3 7E 35 A1 80 F8 F8 AA 97 D7 18 18 AC E5 A8 91 99 2C A2 38 F3 59 6 6D C4 93 18 C9 EA 52 65 76 9F 26 A1 2D 9A 48 77 21 95 F3 54 26 F5 AD 11 93 25 9 9E AA B4 9C 1C 9A 24 14 9D F9 AA 57 12 73 41 2D 10 F9 9F 2F 1A 50 E7 15 24 5C 9E 39 8 E9 57 E1 B9 39 A4 D3 B8 [... cut ...]  8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8 8

UCTRONICS, 2018-09-14:
@juicejams 
Please use the short line to connect the camera to the board

juicejams, 2018-09-17:
@UCTRONICS Thank you so much for all the help. It is working now. 
There should be better conductive jumper DuPont cables for hobbie kits like copper wires or even gold so there won't be signal loss for long cables. Not sure if there's a market for it but I'd definitely buy better quality ones after this experience.

closed 2018-09-17 by juicejams