ArduCAM/Arduino #362: Over Size Error OV5642 and ESP8266 NodeMCU

inquisitivebug, 2018-09-27:
Hello there,
I am using a ESP8266 NodeMCU with the Arducam Mini Module Camera Shield 5MP Plus OV5642 Camera Module. I looked couple of different Arducam INO files on Github and on Instructables and tried to adapt the code and upload it into the NodeMCU. Regardless of the sketches I tried, when I upload the code onto the device, it is successful. I get the IP address and the Device started message. However, I am not able to get the "IP address/capture" and "IP address/stream" to work.

When attempting IP address"/capture" in the browser, I instantly get an "Over Size" error. I noticed on another thread from 2016 that talked about compression. Not sure what setting I modify for compression or resolution.

https://github.com/ArduCAM/Arduino/issues/66

In it, ARDUCAM said the buffer issue would be fixed in Arducam Mini Module Camera Shield 5MP Plus.

I looked at the part of code which revolves around "Over Size".

In one of the Ino files I tried, there is a MAX_FIFO_SIZE variable.

  size_t len = myCAM.read_fifo_length();
  if (len >= MAX_FIFO_SIZE){
    Serial.println("Over size.");
    return;
  }else if (len == 0 ){
    Serial.println("Size is 0.");
    return;
  }

In another Ino file, which was targeted more towards the 2 MP camera the code states that I tried modifying for the 5 MP camera, it states "0x07ffff". Not sure what I need to change this value to.

if (len >= 0x07ffff) {
    Serial.println("Over size.");
    return;
  } else if (len == 0 ) {
    Serial.println("Size is 0.");
    return;
  }

I even tried commenting out this part of the logic, but then I didn't get an image output on the browser. Just a bunch of illegible characters.

When I tried the "IP address/stream", the device crashed and reset instantly.

Any thoughts on how I can make this work? Anywhere else in the code I tweak the resolution or compression to make this work?

UCTRONICS, 2018-09-28:
@inquisitivebug 
Please change the 0x7ffff to 0x7fffff

inquisitivebug, 2018-10-03:
Update: I tried this but it didn't work. I instead purchased a ESP8266 UNO board and tried a different ARDUCAM library. The example "ArduCAM_ESP8266_UNO_Capture" worked for me! I was able to /capture and /stream.

closed 2018-10-03 by inquisitivebug