ArduCAM/Arduino #364: ARDUCAM not being recognized at all

edgargdl, 2018-09-29:
Issue: My arduino Camera even tho is connected as per instructions, is not being recognized at all, like is the same behavior if I have it connected or not.

At this point i am really thinking is broken.

The HUB is not giving any error, just stays in Com Open.

I have a set of warnings when I try lo load the software, i dont think is a killer, but if you could see it is below.. I am also adding a photo of my pin connected:

**Program Name:
ArduCAM_Mini_Video_Streaming**

Arduino: 1.8.6 (Windows 10), Board: "Arduino/Genuino Uno"

ArduCAM_Mini_Video_Streaming:36:1: error: narrowing conversion of '240' from 'int' to 'char' inside { } [-Wnarrowing]

 };

 ^

ArduCAM_Mini_Video_Streaming:36:1: error: narrowing conversion of '196' from 'int' to 'char' inside { } [-Wnarrowing]

ArduCAM_Mini_Video_Streaming:36:1: error: narrowing conversion of '196' from 'int' to 'char' inside { } [-Wnarrowing]

ArduCAM_Mini_Video_Streaming:36:1: error: narrowing conversion of '248' from 'int' to 'char' inside { } [-Wnarrowing]

ArduCAM_Mini_Video_Streaming:36:1: error: narrowing conversion of '224' from 'int' to 'char' inside { } [-Wnarrowing]

exit status 1
narrowing conversion of '240' from 'int' to 'char' inside { } [-Wnarrowing]

Invalid library found in C:\Users\edgar\OneDrive\Documentos\Arduino\libraries\OV7670FIFO: C:\Users\edgar\OneDrive\Documentos\Arduino\libraries\OV7670FIFO
Invalid library found in C:\Users\edgar\OneDrive\Documentos\Arduino\libraries\OV7670FIFO: C:\Users\edgar\OneDrive\Documentos\Arduino\libraries\OV7670FIFO

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

[image]
[image]
[image]

edgargdl, 2018-09-29:
[image]

edgargdl, 2018-09-29:
Ok not sure what happened now I am geting the  SPI interface Error!  ( I just disconnected and connected the cable and changed to be verbosed compiling not sure if that did something)

edgargdl, 2018-09-29:
IS working now haha wiring issue, thanks

UCTRONICS, 2018-09-29:
@edgargdl 
Great to hear you have solved your problems.

edgargdl, 2018-09-29:
Do you havea tutorial of sending this through the web?

UCTRONICS, 2018-09-30:
@edgargdl 
The UNO board doesn't have a wifi module. If you want to send image data through the web with HTTP transmission. I advise you use our Arducam ESP8266 UNO or ArduCAM ESP32 UNO board. We have some demos about how to send data through wifi.
Please refer to the video demo from https://www.youtube.com/watch?v=n1dDGNpbxGM&feature=youtu.be

edgargdl, 2018-10-01:
Thanks for that response, ok what about this, i want to create an app that receives this from the serial port  ( as it does right now,) and then do something with that stream of information,  would like to use java,  but I guess would be easy to do C++ to use the libraries that come with the product, the thing is, i cna not really find any example, the only thing that you guys used this data tois the host app, i would like to make a small application similar to that, and from there i can start sending it https via rest services or something of that sort.

Thanks

elvinosmanov, 2019-06-12:
How did you solve the problem?
I have the same issue