ArduCAM/Arduino #365: ArduCAM Mini OV2640 2MP -Image color change

sidathaw, 2018-09-30:
Hi 
I have a Arducam mini shield OV2640 -2MP camera ,The picture quality is pretty good .But when ever I power down and power up the camera .The first picture color is way off .To bring the image to proper color i have to take 3-4 snapshots.

This gets worst when i try to do the power saving mode:
myCAM.set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);
myCAM.clear_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);

This is my came initialization:
I prefer to keep the below saturation and night mode settings since the image looks better.

 myCAM.set_format(JPEG);
    myCAM.InitCAM();
    myCAM.OV2640_set_JPEG_size(OV2640_800x600);
    delay(1000);
   myCAM.wrSensorReg8_8(0xff, 0x00); //Saturation level 2
    myCAM.wrSensorReg8_8(0x7c, 0x00);
    myCAM.wrSensorReg8_8(0x7d, 0x02);
    myCAM.wrSensorReg8_8(0x7c, 0x03);
    myCAM.wrSensorReg8_8(0x7d, 0x58);
    myCAM.wrSensorReg8_8(0x7d, 0x58);
    delay(1000);
    myCAM.wrSensorReg8_8(0xff, 0x01); //NIGHT MODE
    myCAM.wrSensorReg8_8(0x11, 0x07);
    delay(1000);
  myCAM.clear_fifo_flag();
   delay(1000); 

As you can see I gave about 1 sec delay thinking that it will give enough time to the sensor to recover but still the first two snap shots are way too off the color .Ex: 
This is the first snap shot after power on:
[image]

This is the picture color after 3 snapshots (Real background color)
[image]

I need either to use power saving mode or power on /off the camera for my project since its going to run from solar power and battery operated.But if the image background color is not consistent as i mentioned above i wont be use the Power save mode .
Is there any way to solve that issue ? At the initialization part of the code ?
Appreciate if someone can help me to solve this issue .
Thanks!

UCTRONICS, 2018-10-07:
@sidathaw 
Hi,
Sorry for my late reply because of my China National Day. 
A delay of 2-3 seconds is required after awakening from a low-power state.

sidathaw, 2018-10-07:
> @sidathaw
> Hi,
> Sorry for my late reply because of my China National Day.
> A delay of 2-3 seconds is required after awakening from a low-power state.

Hi Lee thank you ,but i already set the delay to 3 sec the image was yellowish and when i set the delay to 12 seconds I get the image with proper color. Do you think my code delay setting are wrong or its still because of the sensor need 12 seconds to recover ? I tried with 3 cameras and all having the same issue. I need to enable the power saving since its going to be solar powered .If the issue cannot be solved  then we will not be able to use this camera for our iot solar project that requires 200+ cameras.I am totaly fine with a 3 sec delay or maximum 5 seconds delay ,but if its 10-12sec delay that will be unacceptable .Hope you can go the below code and see what went wrong thanks again.

This is the setup code : 
``{
  uint8_t vid,pid;
  uint8_t temp;
  Wire.begin();
  IPAddress myIP = WiFi.localIP();
  Particle.publish("IP:",String(myIP));
  pinMode(SPI_CS, OUTPUT);
  SPI.begin();
  while(1) {
    myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
    temp = myCAM.read_reg(ARDUCHIP_TEST1);
    if(temp != 0x55)
    {
Particle.publish("SPI Error");
    }
    else {
      Particle.publish("SPI ok");
      break;
    }
    Particle.process();
  }
delay(1000);

while(1){
  myCAM.wrSensorReg8_8(0xff, 0x01);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
  if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
    delay(1000);
    continue;
  }
  else {
    Particle.publish("Sensor Module OK");
    break;
  }
}

 delay(100);
    myCAM.set_format(JPEG);
    myCAM.InitCAM();
    myCAM.OV2640_set_JPEG_size(OV2640_800x600);
    delay(1000);
   myCAM.clear_fifo_flag();
   delay(2000);
myCAM.set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);  // Enable power saving mode 
   webServer.begin();
   delay(500);
}
``
This is the capture function of the camera 

void camcapture() {

myCAM.clear_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK); // disable power saving mode
delay(12000); 
    myCAM.flush_fifo();
    myCAM.clear_fifo_flag();
    myCAM.start_capture();
    unsigned long start_time = millis(),
                  last_publish = millis()
    delay(100);
    if(myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))
    {
        myCAM.set_fifo_burst();
        tx_buffer_index = 0;
        temp = 0;
        while (bytesRead < length)
        {
            temp_last = temp;
            temp = myCAM.read_fifo();
            bytesRead++;;
            buffer[tx_buffer_index++] = temp;
            if (tx_buffer_index >= TX_BUFFER_MAX) {
                client.write(buffer, tx_buffer_index);
                tx_buffer_index = 0;
                Particle.process();
            }
            if (bytesRead > 2048000) {
                break;
            }
        }
    }
    client.flush();
    client.stop();
    delay(500);
 myCAM.set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);   //Enable power saving mode 
}

UCTRONICS, 2018-10-08:
@sidathaw 
It has a lot to do with the camera frame rate and AWB mode.
I advise you close the AWB mode and configure the parameters manually.
When the camera's image is normal, remember the parameter values of 0xCC 0xCD 0xCE.
Then close the AWB mode and write the values manually.
[image]

sidathaw, 2018-10-08:
Thanks you for your quick response ,Unfortunately I tried your method and still its giving the yellow pale image .I don't know why it needs 10-12 second to fully recover the image sensor(to get the proper color) Just to mention without power down mask the image is perfectly fine.I wonder have you ever had this experience before.If not i think there is nothing else we can do unless giving a 12 sec delay after sensor awake to capture image.Thanks again !