ArduCAM/Arduino #366: OV2640 with Raspberry Pi 3B+

pranjalranjan299, 2018-10-03:
What is the procedure to use OV2640 with Raspberry Pi 3B+? Can you help me with the pin diagram as well as the various codes? I would like to use the camera to click some pictures and save it to the Pi.

UCTRONICS, 2018-10-07:
@pranjalranjan299 
Please refer to https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/RaspberryPi

Medacher, 2019-04-01:
I cant find  any pin diagramm in the link u gave i tried everything in that project but when it save the picture i find an empty picture that contains no data
I think the software is going well. Its just the  pin diagram is not given

UCTRONICS, 2019-04-02:
@Medacher 
We have updated the code here https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera