ArduCAM/Arduino #367: STM32F4 with Arducam V2

OldMartinThom, 2018-10-11:
I am trying run the V2 with an STM32F429I processor on an ST neucelo dev board. The h/w appears to be working individually for the SPI bus to the V2 chip, LCD controller and the SD card. The sketch reports that the SD card is present, "SD Card detected!" appears, SPI reports "SPI interface OK!".
But when I try the V2 touch example which sets up the SPI bus to the V2 chip, the SPI to the SD card I find that setting up the SD card conflicts with the SPI to the camera (they have seperate pins for the CS lines as usual).

Any ideas?

Regards
Martin

UCTRONICS, 2018-10-12:
@OldMartinThom 
Have you tested the ArduCAM_Shield_V2_Video_Streaming demo?
Firstly, ensure the camera can work normally without lcd and touch.
Then try the Playback demo to test the lcd and camera.
If they can work normally, OK, the problem is on the touchscreen control.
I advise you test the touchscreen using touchpaint.ino demo which is under 
..\Arduino\libraries\ArduCAM_Touch\examples\touchpaint path.

OldMartinThom, 2018-10-21:
Thanks for the ideas.
I found that without setting up the touch system, there is still a conflict between the SD card and the V2 chip. But by adding SPI.begin(); after writing to the SD card (or any other SD card activity), the SPI to the V2 chip works.
I will try the Touch interface at a later date.
Regards
Martin