ArduCAM/Arduino #368: Build errors (lots of) with Teensy 3.x (Arduino IDE 1.8.7 + Teensyduino 1.44)

XFer012, 2018-10-12:
Hello,
just purchased an Arduino Mini Module 2MP to use with my Teensy 3.2 and 3.6 boards.

I'm using your latest library (https://github.com/ArduCAM/Arduino) and trying to build this example
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_2MP_OV2640_functions

I already edited "memorysaver.h" to define (only) the 2mp Mini module.

I use Arduino IDE v1.8.7 and latest Teensyduino add-on v1.44, but when I select a Teensy 3.2 or 3.6 board and try to build the sketch, I get many errors, for example:

`In file included from d:\temp\arduino_modified_sketch_232629\ArduCAM_Mini_2MP_OV2640_functions.ino:11:0:

F:\Arduino\libraries\ArduCAM/ArduCAM.h:713:2: error: 'regtype' does not name a type

  regtype *P_CS;

  ^

F:\Arduino\libraries\ArduCAM/ArduCAM.h:714:2: error: 'regsize' does not name a type

  regsize B_CS;

  ^

ArduCAM_Mini_2MP_OV2640_functions: In function 'void loop()':
ArduCAM_Mini_2MP_OV2640_functions:471: warning: unused variable 'temp_last' 
     uint8_t temp, temp_last;

                   ^

Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 
Using library ArduCAM in folder: F:\Arduino\libraries\ArduCAM (legacy)
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI 
Error compiling for board Teensy 3.2 / 3.1.
`

It looks like the patches proposed by "sumotoy" to get Arducam library work with Teensy were reverted, and Teensy does not work anymore?

UCTRONICS, 2018-10-13:
@XFer012 
ArduCAM library has not added support for Teensy 3.2 and 3.6 boards.  Please tell me more information about the Teensy 3.2 and 3.6 boards and I will help you add support for it.

XFer012, 2018-10-13:
Thanks for your support.
Teensy is a very powerful platform, way faster then any Arduino (and with much more RAM and fast SD interface), so it's quite interesting for camera-based projects!
It is an ARM mcu board which works with Arduino IDE after installing "Teensyduino":
https://www.pjrc.com/teensy/
https://www.pjrc.com/teensy/teensyduino.html

Time ago, user "sumotoy" already contributed patches here, to have Teensy 3.x work with Arducam:

https://github.com/ArduCAM/Arduino/pull/7

but looks like they were ignored. :(

Just updating those patches to current source should do the trick; "sumotoy" even started a branch to work on Arducam library for Teensy but now won't compile anymore (too many changes):
https://github.com/sumotoy/ArduCAM

UCTRONICS, 2018-10-15:
@XFer012
Hi,
I have updated the ArduCAM library code to support teensyduino , please download the new library to try.

UCTRONICS, 2018-10-15:
@XFer012 
What's the main chip type of your teensy and I plan to buy one.

XFer012, 2018-10-15:
Wow, that's great, thanks a lot !!! :) :) :)
Will try and report back. :)

Teensy is a actually family of dev boards.
The cheapest board is Teensy LC and its main chip is a NXP Kinetis MKL26Z64VFT4 (ARM Cortex M0+)
The "middleweight" is Teensy 3.2; main chip is NXP Kinetis MK20DX256VLH7 (ARM Cortex M4)
The most powerful is Teensy 3.6, with MK66FX1M0VMD18 (ARM Cortex M4 with hw floating point)
On Teensy boards, there's a companion chip as a bootloader: 
https://www.pjrc.com/store/ic_mkl02.html

XFer012, 2018-10-15:
Hello,
I've installed your latest library and tried again, on Teensy 3.6, 
"ArduCAM_Mini_2MP_OV2640_functions.ino"
using Arduino IDE 1.8.5, Teensyduino 1.44 

This time the sketch builds correctly :) but unfortunately I get this error:

**ACK CMD ArduCAM Start! END
ACK CMD SPI interface Error! END**

I've attached SPI pins to Teensy SPI#0 (default SPI), which means:
MOSI on pin 11
MISO on pin 12
SCK on pin 13
and CS is pin 10 (configured appropriately in the sketch: const int CS = 10)

Anything I may try?
Thanks for your support!

XFer012, 2018-10-20:
I purchased a logic analyzer.
This is the track I sampled:
[image]

XFer012, 2018-10-20:
Sorry: faulty jumper. Works OK with Arduino Uno (not yet with Teensy 3.6).

XFer012, 2018-10-20:
New (clearer?) capture from Teensy 3.6.
Overall from start:
[image]

Zooming in on bus event around 3200ms:
[image]

XFer012, 2018-10-21:
Resolved!

Sorry I can't send a patch file now (not to mention a pull request), but the correction is very easy:
Arducam.cpp
Line 133:
```
- 		#if (defined(ESP8266)||defined(ESP32))
+ 		#if (defined(ESP8266)||defined(ESP32)||defined(TEENSYDUINO))
```

XFer012, 2018-10-21:
It would be great to also add SPI transaction, as suggested in "sumotoy" patches

XFer012, 2018-10-21:
With this patch for current Arducam.cpp, Teensy error is fixed and SPI transactions are automatically supported (with compatible SPI libraries). 
Backward compatible with SPI libraries without transactions support.
All thanks to "sumotoy", I just updated his patches.

Sorry I'm not able to issue a proper git pull request:

[arducam_cpp_patch_20181021-01.txt](https://github.com/ArduCAM/Arduino/files/2498998/arducam_cpp_patch_20181021-01.txt)

KurtE, 2018-11-13:
@XFer012 and @UCTRONICS 

The current version does not build on a Teensy. 

That is the last || TEENSYDUINO at about line 3226 causes the issue where it then tries to use ESP specific register names which are not defined for Teensy. 

That is it should go back to
```
#if defined(ESP8266)
inline void ArduCAM::setDataBits(uint16_t bits) {
  const uint32_t mask = ~((SPIMMOSI << SPILMOSI) | (SPIMMISO << SPILMISO));
  bits--;
...
```

UCTRONICS, 2018-11-14:
@KurtE 
Thank you very much.
I will optimize it now.