ArduCAM/Arduino #370: Empty JPG Images on Raspberry Pi Zero & 3b

martinlang83, 2018-10-19:
Hello,

as written in the topic, i dont get any images from my Arducam Mini 2MP. There are also two older ISSUES describing the same problem but didn't get fixed.
I tested it with two new camera modules and with a PI Zero and also PI 3 b. It is always the same error.

[image]

in the image it is to see that the code don't run through all states. there is no jpg header sent by the cam. when i look the MISO line with my oscilloscope I also see that the cam only replys 0x00 on the line.

I installed the driver like described in the readme.
I think there is an error in this driver for some reason.

can someone help?

regards

martinlang83, 2018-10-20:
any ideas? i think there is a bug in this driver implementation for raspberry. can i provide more information in order to help?

ArduCAM, 2018-10-21:
Please use the source code from: https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera

martinlang83, 2018-10-21:
@ArduCAM thank you really much for your help. unfurtunatly even this lib does not work really. I installed it like described in the Readme /Quick Install instruction. However when I run "sudo ./run_Arducam_Demo" nothing happens. no output, no open port 2002.

can you help again?

[image]

martinlang83, 2018-10-21:
@ArduCAM I'm using pi zero and figured out that pi zero has the BCM2835, in this lib there is a base address defined for BCM2836. I changed the base periph address define to the 2835, however now I'm seeing this errors:

[image]

martinlang83, 2018-10-21:
okay I found the mistake. in the lib the CS Pin is defined to 17 and i connected it to GPIO 8. changed it solved the interface error.

closed 2018-10-21 by martinlang83