ArduCAM/Arduino #371: Cannot find MISO or MOSI pins on camera module

ironfist96, 2018-10-22:
I'm trying to connect the camera (OV2640) to the Arduino mega2560, and I cannot understand which is pin is to be connected and where. The camera module that I have looks a bit different from what the others have and as such, I wasn't able to find a post with a similar problem.
[image]

As you can this module doesn't have a MISO, MOSI, and the other pins of the like, so can you please help me out with the pin configurations?

[image]

UCTRONICS, 2018-10-23:
@ironfist96 
This kind of camera should be used with our Arducam shield V2 board.
If you want to use SPI camera, please refer to the https://www.amazon.com/Arducam-Module-Megapixels-Arduino-Mega2560/dp/B012UXNDOY/ref=sr_1_3?m=A2IAB2RW3LLT8D&s=merchant-items&ie=UTF8&qid=1540259687&sr=1-3&refinements=p_4%3AArducam

ironfist96, 2018-10-23:
@UCTRONICS
Thanks for the info, but is there any way I could use the this camera module without the arducam shield? Time is of the essence for my project and finding the shield and shipping it will take time.

UCTRONICS, 2018-10-23:
@ironfist96 
There are no other ways to use this module on mega2560.
I advise you replace a camera.

ironfist96, 2018-10-23:
@UCTRONICS Ah gotcha. I guess I'll be ordering an SPI Camera module. Thanks for the help.

closed 2018-11-01 by ironfist96