ArduCAM/Arduino #372: Take Photo When Ultrasonic sensor reads below 10cm (OV2640)

nanz17, 2018-10-31:
I want the camera module to take a photo when the sensor reads less than 10 cm. I'm currently using example Capture2SD code but I want the photo to be taken automatically within the code, not through the host app. Here is the code for the ultrasonic sensor.
// ---------------------------------------------------------------------------
// Example NewPing library sketch that does a ping about 20 times per second.
// ---------------------------------------------------------------------------

#include <NewPing.h>

#define TRIGGER_PIN  12  // Arduino pin tied to trigger pin on the ultrasonic sensor.
#define ECHO_PIN     11  // Arduino pin tied to echo pin on the ultrasonic sensor.
#define MAX_DISTANCE 200 // Maximum distance we want to ping for (in centimeters). Maximum sensor distance is rated at 400-500cm.

NewPing sonar(TRIGGER_PIN, ECHO_PIN, MAX_DISTANCE); // NewPing setup of pins and maximum distance.

void setup() {
  Serial.begin(115200); // Open serial monitor at 115200 baud to see ping results.
}

void loop() {
  delay(50);                     // Wait 50ms between pings (about 20 pings/sec). 29ms should be the shortest delay between pings.
  Serial.print("Ping: ");
  Serial.print(sonar.ping_cm()); // Send ping, get distance in cm and print result
 if (sonar.ping_cm()<10);
{ //Take Picture and save to SD card}

  Serial.println("cm");
}

UCTRONICS, 2018-10-31:
@nanz17
 It is easy to implement.
You  can refer to the https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Capture2SD/ArduCAM_Mini_Capture2SD.ino

nanz17, 2018-10-31:
Thanks but that's the example code. It's hard to understand how exactly I can implement it without the host app.

UCTRONICS, 2018-11-01:
@nanz17 
That demo does not use the host V2 app. What's the function of the ArduCAM_Mini_Capture2SD.ino is getting the image and store the image to the SD card periodically.

nanz17, 2018-11-01:
Okay thanks. I got it :).

KlydeLencioco, 2018-11-29:
@nanz17 have you done this one? How did you connect your ultrasonic pins? Is there are available pins when using the arducam shield?