ArduCAM/Arduino #373: Video2SD demo: invalid AVI file and capture speed is wrong

XFer012, 2018-11-01:
Hello, using your demo code with my Arducam_Mini_2MP
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_Video2SD

1) The saved clip is an incorrect AVI (MJPEG) file: most players refuse to play it ("malformed header" and/or "invalid chunks"). 
With the few players which nevertheless try to play the clip, video speed is wrong (way too fast, like 100+ FPS) and only 100 out of 200 frames are decoded (the others are skipped!).
Unfortunately it's too difficult for me to understand what's going on; maybe chunk headers are wrongly calculated/written?

2) Despite having capture rate at "0x05" (5 FPS), the camera does not record for 40 seconds (200 frames / 5 fps), but for 14 seconds only. 
This suggests actual capture rate is wrong...?

Thank you very much for your support.
Your products are wonderful, they warrant just a bit more effort on the software/docs side maybe? :)

UCTRONICS, 2018-11-02:
@XFer012 
Hi, This is just a test demo. We just use it to test our camera. We fake an AVI header and write the image data to the AVI file. For Arduino UNO platform, it is very difficult to encode the AVI file.
If you are interested, we are willing to work together with you.

XFer012, 2018-11-02:
Thanks for the explanation :)

Yes I'm interested in collaborating to have a working AVI! 
Just let me know how to help out.

I use the Teensy 3.6, which is very powerful and has no problems writing video data to microSD in realtime (write speed is up to 12-16 MB/s); in fact, I'm getting sustained 15 FPS at 320x240 right now (albeit the final AVI is broken as said).

XFer012, 2018-11-02:
As a first step, this is my modified version of your demo code.
I capture and write video frames as fast as possible, and only then I compute the actually attained FPS and frame duration and update the AVI header appropriately. 
Still, 50% of video chunks are rejected by video players. :(

[ArduCAM_Mini_Video2SD_FC.zip](https://github.com/ArduCAM/Arduino/files/2542370/ArduCAM_Mini_Video2SD_FC.zip)

UCTRONICS, 2018-11-03:
@XFer012 
Maybe it still exist some problems for the AVI header. You can use VLC media player to try .

XFer012, 2018-11-03:
Same thing. Half the frames are dropped because of "video chunk error".

Looks like the main AVI header (first 240 bytes) is OK, but the video segments and/or their "mini-headers" may be wrong. 

Wild guess: video chunk headers report wrong chunk size?

Is it possible for you to have a look at it? In your code, there are too many things I don't understand. :(

XFer012, 2018-11-05:
Resolved!
The chunk size math is wrong: you don't have to add "remnant" to jpeg_size (and movi_size).
Moreover, there's no need to align jpeg to 4 bytes: aligning to 2 bytes is legal, so you can simply do:
remnant = jpeg_size & 0x00000001

Tomorrow I'll clean my code a bit and will attach the corrected .ino here. :)

I get a solid 14 FPS at 320x240; honestly I hoped for more, but looks like camera->FIFO->CPLD has a bandwidth bottleneck...?

UCTRONICS, 2018-11-06:
@XFer012 
Great to hear you have solved the bug. Looking forward to testing your new code.
I think we will have more cooperation on the Arduino platform.

XFer012, 2018-11-06:
This is the generic, "Arduino-friendly" version:

[ArduCAM_Mini_Video2SD_mod.zip](https://github.com/ArduCAM/Arduino/files/2554972/ArduCAM_Mini_Video2SD_mod.zip)

XFer012, 2018-11-06:
And this is the Teensy 3.5/3.6 optimized version (uses onboard microSD slot in SDIO mode, SdFat library,
[ArduCAM_Mini_Video2SD_T36.zip](https://github.com/ArduCAM/Arduino/files/2554981/ArduCAM_Mini_Video2SD_T36.zip)
 larger buffer)

UCTRONICS, 2018-11-07:
@XFer012 
Great, I will test it and update our code. Thank you very much.

UCTRONICS, 2018-11-07:
@XFer012 
Hi, I have tested your code and I get this message
[image]
And the size of the avi file just 1K. I don't know why.

XFer012, 2018-11-07:
Wow this is strange.
I don't have an Arduino board (I work exclusively on Teensy), so it's quite difficult for me to reproduce the issue.
I'll see if I can borrow an Arduino Uno.

XFer012, 2018-11-07:
Hello,
on Arduino, I can't have SD and Camera working at the same time (I have Camera CS on 7 and SD CS on 4).
I keep getting SD Error: like if SD card module does not get selected on the bus.
If I disconnect the Camera, the SD works.

If I disconnect the SD (and comment out the relevant code), I receive video correctly:

`
ArduCAM Start!

SPI interface OK.
OV2640 detected.

Recording 200 video frames: please wait...

*** Video recorded and saved ***

Recorded 13s in 200 frames
File size is 1185486 bytes
Actual FPS is 14.29
Max data rate is 82815 byte/s
Frame duration is 69975 us
Average frame length is 5915 bytes

`

XFer012, 2018-11-07:
So my guess is, somehow Camera and SD don't play well together on Arduino (work on Teensy because of totally different SD bus, SDIO-type with different pins, separate clock etc.).

UCTRONICS, 2018-11-08:
@XFer012 
Hi,
I have many platforms such as Arduino UNO ESP8266 UNO ESP32 UNO.
I can send them to you for free and you can test it on them, please tell me your address and your contact information.

XFer012, 2018-11-08:
@UCTRONICS thanks for the offer, but I live in Italy, it would be a long journey (and quite the shipping cost, not to mention customs issues).
Let me see if I can borrow another Arduino, this time a genuine one, with separate SD reader/shield (my test was on a Fishino Uno with integrated SD reader).
But I think something happens with the SPI CS line of ArducamMini2MP, which prevents SD reader to work... will try with the logic analyzer.

XFer012, 2018-11-08:
@UCTRONICS The Mini2MP module sometimes does NOT release CS :( , thus SD writes are impossible.
A power cycle is then needed. :(

Now I rearranged setup() on the Arduino version, and problems are less frequent, but still present on occasion (Mini hardware or CPLD firmware problem, I think).

New version:
[ArduCAM_Mini_Video2SD_mod.zip](https://github.com/ArduCAM/Arduino/files/2562063/ArduCAM_Mini_Video2SD_mod.zip)

Output:
Opening port
Port open
SD Card detected.
SPI interface OK.
OV2640 detected.

File name will be 372.avi

Recording 200 video frames: please wait...

*** Video recorded and saved ***

Recorded 22s in 200 frames
File size is 714370 bytes
Actual FPS is 8.95
Max data rate is 32038 byte/s
Frame duration is 111790 us
Average frame length is 3559 bytes

XFer012, 2018-11-12:
@UCTRONICS excellent, let me know.
BTW, I would also like to try/adapt Arducam Shield v2 to non-RPi ARM boards, like Rock64 and Odroid C2 (I already have both): could we have a private chat about that?

UCTRONICS, 2018-11-13:
@XFer012 
Great, I have tested and it can work normally. You are right. If the SPI bus is busy, the SD card will not be written.  In our Arducam driver, each read or write register, we will set the CS high. 
About the Arducam shield V2, I have not tested on ARM board, but I have written a demo using Arducam mini camera on the RPI board. You can get it on https://github.com/ArduCAM/RaspberryPi/tree/master/SPI_Camera

XFer012, 2018-11-13:
@UCTRONICS feel free to include this sample code in your repository, if you think it's worth.
About Arducam Shield V2: I've seen your RPi demo is GUI-based and quite RPi-specific.
I think the Shield V2 could be very interesting for specific ARM board application, like scientific/technical unattended image acquisition (headless, no GUI).
Is it possible to have a V2 Shield on loan, so I could test (and write non-GUI support software) it on Odroid and Rock64 ARM boards? Of course I would share the software afterwards. I already have a few (Shield-supported) cameras to work with, but no Shield.

KaiserGagius, 2023-10-12:
Hello, it´s been quite a while but I´ve been reading Microsoft´s documentation and I still don´t understand how the header was constructed, isn´t it supposed to be 13 32bit segments (52 bytes)? How on earth did you end up with 240 bytes?

My main issue is that the fps are far too low, I´m unable to get it above 7fps though it might just be an issue with the clock (changing the SPI clock divider to 2 changes nothing). 

Maybe there is a way to save a smaller RAW video so no time is lost encoding but I´m at a loss here.