ArduCAM/Arduino #374: Chip Select Line Staying Low After Capture and Interfering with Other Devices

IoTPanic, 2018-11-06:
## Setup
This issue was found utilizing the OV2640 Mini module, as well as an Ardunio Zero clone using copied and pasted example code. There is a flash storage chip as well as a SX127X LoRa radio utilizing the same bus.
## Symptoms
When this issue occurs, the Arduino will crash while executing an SPI transfer as used during photo capture on lines such as `while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));` and `temp =  SPI.transfer(0x00);`. Crashing being defined as locked up, no longer executing further code.
## What Occurs
On the capture before the crash, using an Oscilloscope we noticed the chip select line not returning back to high after capture randomly (line stays low, the camera continues to believe to be the active device on the SPI bus), then while transceiving data with the LoRa radio, some information is incorrect. During the next capture, the `temp = SPI.transfer(0x00)` line will crash. Still trying to diagnose why `while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));` crashes the Arduino. 

The issue seems to arise with a pattern of utilizing the radio and camera in a specific pattern, have not figured out the pattern, but it has been a noticeable issue for a while, it seems random when just trying to diagnose with no tools. I have seen others online complain about soundingly similar issues.
## Code Snippet
```
void capture() {
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  bool is_header = false;
  File outFile;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println(F("start Capture"));

  while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK)); // CAN CRASH HERE

  Serial.println(F("Capture Done."));

  length = myCAM.read_fifo_length(); // CAN CRASH HERE

  Serial.print(F("The fifo length is :"));
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //384K
  {
    Serial.println(F("Over size."));
    return ;
  }
  if (length == 0 ) //0 kb
  {
    Serial.println(F("Size is 0."));
    return ;
  }
  //Construct a file name
  k = k + 1;
  itoa(k, str, 10);
  strcat(str, ".jpg");
  //Open the new file
  outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
  if (!outFile) {
    Serial.println(F("File open faild"));
    return;
  }
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();
  while ( length-- )
  {
    temp_last = temp;

    temp =  SPI.transfer(0x00); // CAN CRASH HERE

    //Read JPEG data from FIFO
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    {
      buf[i++] = temp;  //save the last  0XD9
      //Write the remain bytes in the buffer
      myCAM.CS_HIGH();
      outFile.write(buf, i);
      //Close the file
      outFile.close();
      Serial.println(F("Image save OK."));
      is_header = false;
      i = 0;
    }
    if (is_header == true)
    {
      //Write image data to buffer if not full
      if (i < 256)
        buf[i++] = temp;
      else
      {
        //Write 256 bytes image data to file
        myCAM.CS_HIGH();
        outFile.write(buf, 256);
        i = 0;
        buf[i++] = temp;
        myCAM.CS_LOW();

        myCAM.set_fifo_burst(); // CAN CRASH HERE

      }
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      buf[i++] = temp_last;
      buf[i++] = temp;
    }
  }
}
```

UCTRONICS, 2018-11-07:
@IoTPanic 
Thanks for your detail description. I have checked the code you attached.
After the SPI read  0xff 0xd9 which is the end of the jpeg data, it will make the CS high.
If it can't read the end of the jpeg data, it will maintain the CS in the LOW state. 
You can add  myCAM.CS_HIGH(); after the length-- =0.

IoTPanic, 2018-11-07:
@UCTRONICS Yeah, so after literally days of debugging, I have figured out the issue. The issue is on line `length = myCAM.read_fifo_length();`, where either the ArduCAM OV2640 Mini or ArduCAM library produces an incorrect FIFO length result. (As described in http://www.oz4.us/2017/03/arducam-too-unstable-to-be-useful.html I just found) which causes the code to not find the JPEG endbytes (I assume because the returned FIFO length is too small) so then the `while(length--)` loop runs out while keeping the chip select LOW. This can do one of two things, one is it can send data down the MISO line which interferes with other devices on the bus, or crash on the next `SPI.transfer`.

While trying to solve my issue, I have came across these issues, and it sounds like this problem did them no favors. I'm sure if I combed through I could find more. #50  #292  
>"cam_data_len" error means that the received frame data size is less than anticipated it should be. Their might come from different situations, but the USB through put might takes the most important role for that. Please connect the camera to USB root hub only, without other USB device.

This is a major issue, and I have been using your camera for months in development of an IoT device, and have suffered from lots of corruption on my radio, random crashes, and a host of other issues.

UCTRONICS, 2018-11-07:
@IoTPanic 
Hi, 
Yes, many factors will make the length error. such as the camera can't init normally and the hardware connection is not stable.etc.
Before shipping, we will test it on Arduino UNO and STM32 platform. Different platform, the time slot is different, so sometimes, we need to set the right time slot to ensure the camera can init normally. If the camera doesn't init OK, which means the camera can't work normally. 
So please test the camera on Arduino UNO platform using our test demo to check if the camera is defective.

IoTPanic, 2018-11-07:
@UCTRONICS So, I have currently four units with ArduCAMs built in, plus an entire box of cameras and I've tested multiple cameras on multiple units, and throughout development, multiple boards were used including boards that are declared as compatible such as the Mega 2650. The camera is initializing just fine, and most of the time produces correct results. it seemingly randomly returns incorrect FIFO lengths. No matter the micro or the specific camera.

UCTRONICS, 2018-11-08:
@IoTPanic 
Please attached me some pictures of your hardware connection. I advise you use the short connection line as short as possible.

IoTPanic, 2018-11-08:
@UCTRONICS 
https://ibb.co/e6qYhq
https://ibb.co/fU6vvA

UCTRONICS, 2018-11-08:
@IoTPanic 
What's this board? Do you make an adapter board by yourself?

IoTPanic, 2018-11-12:
Aha, my bad. So it turns out the LoRa library was doing some interrupt stuff in the background I didn't know about. My bad.

IoTPanic, 2018-11-15:
Okay, so there is an actual issue. But it is not nearly as bad as the one I reported. So what is happening is the Radiohead library in the background uses interrupts (I apologize, it did not mention in documentation that I saw so I looked at the source) so when I receive a LoRa packet at the same time the camera is retrieving the FIFO length, it goes to the interrupt, flips a few flags in the radio, and gets the wrong FIFO size. So my report thus far has been my bad. However, when I run `SPI.beginTransaction(12000000, MSB, MODE_0)` before your code your code stops functioning. I tried using this in the camera setup code and got the classic `SPI Erorr`. These were the SPI settings I was using before, and what it's supposed to do is prevent the interrupt from interrupting in the Arduino framework.  Do you have a solution?

UCTRONICS, 2018-11-15:
@IoTPanic 
Great to hear you have found the problem. You can change SPI.beginTransaction(12000000, MSB, MODE_0) to  SPI.beginTransaction(8000000, MSB, MODE_0).

closed 2018-11-15 by IoTPanic