ArduCAM/Arduino #375: Arducam Mini Fifo oversize

Donovator122, 2018-11-07:
Hello, 
I am programing the Arducam mini OV2640 2mp with Arduino. I am using an Arduino mega board. I am planing on writing to an SD card. I have the library downloaded. I am having issues with the SPI interface and when i comment that section out it is able to detect the camera, start a capture, complete the capture and read the fifo size. However, from this point it reads the fifo size as 8388607 which is oversize. What needs to be changed in order to decrease the fifo size, and make the SPI work?
Thank you,
-Donovan

IoTPanic, 2018-11-07:
Check out #374, you are not alone with FIFO length issues

UCTRONICS, 2018-11-08:
@Donovator122 
You can't comment that which is used to detect the SPI interface.
If the SPI interface is error, anything about SPI operation is meaningless. So, please check the SPI interface firstly. Please attach me your hardware connection and I will help you check it.

UCTRONICS, 2018-11-08:
@Donovator122 
Hi, I have checked your wiring diagram. The SPI signal does not pin 10/11/12  you should connect it to ICSP
[image]

Donovator122, 2018-11-08:
That did the trick, 
the SPI is now working, the fifo file changes number randomly, but is not saving image:

start Capture
Capture Done.
The fifo length is :274
start Capture
Capture Done.
The fifo length is :4

Is this the correct wiring now? What should change to make the file save process complete?

[Minicam:SD.fzz.zip](https://github.com/ArduCAM/Arduino/files/2562302/Minicam.SD.fzz.zip)
[minicam_to_sd.ino.zip](https://github.com/ArduCAM/Arduino/files/2562328/minicam_to_sd.ino.zip)

UCTRONICS, 2018-11-12:
@Donovator122 
I  think the fifo length is too short and it is not right.
Please ensure the connection wire is as short as possible.

UCTRONICS, 2018-11-15:
@Donovator122 
Firstly, please use our videoStreaming demo on UNO board to test the camera. You can get it from https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming/ArduCAM_Mini_Video_Streaming.ino
From the fifo length, I guess the camera has not init OK.

Faisalameen2, 2022-04-15:
> 

Can you show me how you connect the Arducam with Mega board ?