ArduCAM/Arduino #378: Redefined #define preproc warning: pulse_high and pulse_low

kebwi, 2018-11-14:
At least one example script (I didn't check others) generates the following compiler warning. There's a pretty straightforward redefinition of two #define preproc statements in the library. I'm not sufficiently familiar with the library to know the "intended" code organization that would properly fix this.

<pre>
In file included from [...]/Arduino/libraries/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Camera_Playback/ArduCAM_Shield_V2_Camera_Playback.ino:18:0:
[...]/Arduino/libraries/ArduCAM/ArduCAM.h:114:0: warning: "pulse_high" redefined
 #define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
 ^

In file included from [...]/Arduino/libraries/UTFT4ArduCAM_SPI/UTFT_SPI.h:189:0,
                 from [...]/Arduino/libraries/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Camera_Playback/ArduCAM_Shield_V2_Camera_Playback.ino:15:
[...]/Arduino/libraries/UTFT4ArduCAM_SPI/HW_AVR_SPI_defines.h:5:0: note: this is the location of the previous definition
 #define pulse_high(reg, bitmask) sbi(reg, bitmask); UTFT_cbi(reg, bitmask);
 ^

In file included from [...]/Arduino/libraries/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Camera_Playback/ArduCAM_Shield_V2_Camera_Playback.ino:18:0:
[...]/Arduino/libraries/ArduCAM/ArduCAM.h:115:0: warning: "pulse_low" redefined
 #define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);
 ^

In file included from [...]/Arduino/libraries/UTFT4ArduCAM_SPI/UTFT_SPI.h:189:0,
                 from [...]/Arduino/libraries/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Camera_Playback/ArduCAM_Shield_V2_Camera_Playback.ino:15:
[...]/Arduino/libraries/UTFT4ArduCAM_SPI/HW_AVR_SPI_defines.h:6:0: note: this is the location of the previous definition
 #define pulse_low(reg, bitmask) UTFT_cbi(reg, bitmask); sbi(reg, bitmask);
 ^
</pre>

UCTRONICS, 2018-11-15:
@kebwi 
Don't worry and please refer to our operation video to know the detail. 
Referring https://www.youtube.com/watch?v=XMik38TNqGk&feature=youtu.be

kebwi, 2018-11-15:
I don't understand. Are you saying that if I had followed the video's instructions properly the warning would not have appeared. I think I followed it exactly.