ArduCAM/Arduino #379: Low power settings for OV2640

steveww, 2018-11-20:
From the sparse documentation I have found the following registers:

0x86 bit 2 borad LDO
0x86 bit 1 PWDN pin on sensor???
0x03 bit 6 low power mode control
0x05 bit 1, bit 2 sensor power down I/O
0x06 bit 1 sensor power down I/O
0x45 bit 1 sensor power down I/O

Alas this is all the documentation mentions. Please provide more information as to what these registers do and the impact of setting/clearing these bits.

My goal is to be able to put the board to sleep to reduce power consumption while running on batteries, waking up on demand.

UCTRONICS, 2018-11-20:
@steveww 
Write 0x15 to the register 0x06 address to go to sleep mode 
Write 0x17 to the register 0x06 address to go to normal mode

steveww, 2018-11-20:
If this is the case, please improve your documentation to make this clear. The documentation is totally ambiguous on these settings.

What are the power setting on registers 0x05 0x45?