ArduCAM/Arduino #380: What are is the correct method to make a cameraInit() wakeup function

martinberlin, 2018-12-04:
I started to write this ticket in the Arducam ESP32 repository but no one seems to give support there so I will copy it here.
I'm trying to wake up camera only on the moment of taking a picture and then shutting it of via a P-channel mosfet.

So I built [this functions](https://github.com/martinberlin/FS32/blob/master/src/FasarekFS2.ino#L716): 

```
void cameraInit() {
  digitalWrite(gpioCameraVcc, LOW);       // Power camera ON
  // OV5642 Needs special settings otherwise will not wake up
  myCAM.clear_bit(6, GPIO_PWDN_MASK);  // Disable low power
  delay(3);
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  delay(50); 
  myCAM.write_reg(3, 2);               // VSYNC is active HIGH
  delay(47);
}

void cameraOff() {
  digitalWrite(gpioCameraVcc, HIGH); // Power camera OFF
}

```

But I discovered that this times are not working the same with different OV5642 5 megapixels camera. Question is: There is not a more clever way to do this ? Like make a while(check something)
that will return true when camera wakes up fully ?
Can you please reference me to a code snippet ?

UCTRONICS, 2018-12-05:
@martinberlin 
It is due to the auto balance and exposure and we can't get the state of the camera.
It is the best way to add some delay.

martinberlin, 2018-12-05:
Thanks Uctronics, but the problem is that this fixed delay does not work for every OV5642 with the same milliseconds delay. For some I needed to wait up to 800 ms which is way too much. How comes there is no way to do this in a smarter way ?
I cannot tell the developers using my repository to compile 50 times with different values till they find the right delay for their camera. It’s not that it simply does not work, it generates a jpeg that is too big and has no jpeg headers waiting less time than it should. And also there is no way I can find to check this unless I add some kind of condition to check if the image starts as a JPEG and if not there is a second attempt to shoot the picture.
So it’s very hard to make something with battery’s using this camera since consumes a lot and sadly there is no smart way to program a smart wake up that will adjust for every camera. Can’t you please check if there is not any other alternative ?

UCTRONICS, 2018-12-05:
@martinberlin 
You can try to disable the awb function by write the 0x5001 register Bit[0 ] 0:Disable 1: Enable

martinberlin, 2018-12-05:
Thanks. Will try that, but then I won't want automatic white balance, which was not the intention here.  I just want a way to do this cameraIinit() in a smarter way than now. 
That will work for every OV5642 Arducam SPI withouth the need to mess with inconsistent delay() times.