ArduCAM/Arduino #381: Ov5642 produces many times corrupt JPEGs

martinberlin, 2018-12-06:
Using mostly 1600x1200 resolution. Using ESP32 but happens the same with esp8266
https://github.com/martinberlin/FS32 
I’ve a added a p–channel mosfet since I’m making a self made WiFi digital camera and I want to turn it on only on the moment of taking a picture. Information about this is in another issue #380  

The issue is, with this resolution you get a jpeg of 130 to 180 kB size. One of each 5 times, it generates a 480 or more jpeg, that is crap and does not even start like a jpeg. And it’s not because of the turn on/off, it is just unestable. I’m using short cables and all is good soldered so it cannot be the connection. I’m sending arducam one of my cameras so they can try it out there and check this in the laboratory. Arducam has an unique market and it has a potential to be much more, but I must say after reading answers like „sorry we cannot give out the schematics“ and the lack of proper documentation I must say that is not a pleasure at the moment to try to get the most of one of your cameras.
It works, but it is not stable, and it’s not the WiFi. I’m saving also the image on SPIFFs and I know what pictures a badly on the php upload endpoint and what are not.

So I would like to ask:
1.- What is the best way to check the first bytes in C++ and stop the while loop if after reading the first bytes there is no JPEG header? Can you pleas give me a code sample for this ?
2.- Can you please implement a Smart way to check if the camera wake up and is ready to take a picture in next versions? It’s just a no go to make a delay that changes on every different camera (even being both same model OV5642)

UCTRONICS, 2018-12-07:
@martinberlin 
Thanks for your advice.
For the first question, you can see the detail idea and method from the below code.

  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.println(F("ACK IMG END"));
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    break;
    delayMicroseconds(15);
  }
2, At present , we  have not able to get the signal of the  camera wake up successfully.

martinberlin, 2018-12-07:
Great thanks! I will implement that and give you feedback about how it worked out. 
I will be working this weekend in a upload chunker idea, only to be implemented in ESP8266 / ESP32 and any API that can upload chunks and put them together when last arrives, still researching. The idea is to send a JPEG in 20Kb chunks via WiFI with a checksum, and reply with a status 200 (arrived OK) or 400 (arrived false, needs to be resended) to avoid having corrupt JPEGs via WiFi upload. 

@UCTRONICS Point 2- Please keep in mind something like this for future versions. I read about a lot of people using this Arducams with batteries recharged by solar energy, where consumption is KEY, and cannot be that is so hard and there is no acknowledge that the camera is ready. If it's too little time also White balance is affected or JPEG output is corrupted.
It would be a great achievement if in future versions the generated JPEG returns also a checksum so we can check somehow that the received bytes are the same that come out of the Sensor.

UCTRONICS, 2018-12-09:
@martinberlin 
Thanks for your good ideas and we will consider it in careful.