ArduCAM/Arduino #382: Arducam shield v2 LCD cannot display

mobile44, 2018-12-22:
I have no error message. I also check the image in the card to confirm camera working fine. However, the LCD cannot display anything but just bright screen. Following is the serial message...
ArduCAM Start!
SPI interface OK
OV2640 detected.
SD Card detected!
Show Image: 1.bmp
Start Capture
Capture Done.
Display Image...
Image File: 1.bmp
Image File: 2.bmp
Image File: 3.bmp
Image File: 4.bmp
Show Image: 2.bmp
Start Capture
Capture Done.
Display Image...
Image File: 1.bmp
Image File: 2.bmp
Image File: 3.bmp
Image File: 4.bmp
Show Image: 3.bmp
Start Capture
Capture Done.

How can I confirm the LCD without problem? Many thanks!

UCTRONICS, 2018-12-23:
@mobile44 
We have a demo for LCD. have you tried this demo https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Camera_Playback

mobile44, 2018-12-23:
Hi UCTONICS,

I am using ATmega2560 and the camera in ArduCAM Shield V2 is OV2640_CAM. Actually I did test with following default examples. However, I can only prove that camera is able to take picture and store picture in SD card but none of image to be shown on the LCD. Will the LCD come with problem? Thanks!
ArduCAM_Shield_V2_Camera_Playback
ArduCAM_Shield_V2_Capture2SD
ArduCAM_Shield_V2_Digital_Camera
ArduCAM_Shield_V2_Digital_Camera_2560
ArduCAM_Shield_V2_Multi_Capture2SD
ArduCAM_Shield_V2_short_movie_clip
ArduCAM_Shield_V2_Touch
ArduCAM_Shield_V2_Video2SD
ArduCAM_Shield_V2_Video_Streaming

mobile44, 2018-12-23:
Hi UCTONICS,

Referring another article:
http://www.arducam.com/how-to-connect-arducam-shield-to-mega-2560/#more-509

I have also added pinMode(A4, INPUT); & pinMode(A5, INPUT); in setup. But there is no improvement. Do you have any other idea? Thanks!

mobile44, 2018-12-23:
Hi UCTONICS,

Even I change to UNO R3, the situation is same. The LCD is bright but without any image which is same as Mega2560.

UCTRONICS, 2018-12-23:
@mobile44 
Tomorrow I will test it and give you a reply as soon as possible.

mobile44, 2018-12-23:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mobile44, 2018-12-23:
Hi UCTRONICS ,

I have followed the instruction in examples. The UNO R3 should not have pin in wrong position. Hence it is, as if, either the hardware issue or wrong PIN in SPI. Thanks!

UCTRONICS, 2018-12-24:
@mobile44 
Hi,
I have tested and it can work normally. Have you changed another Arduino UNO to try?
If it still exists a problem. Maybe your LCD exist some problems. Please contact our salesman for a replace.