ArduCAM/Arduino #384: OV2640 Mini 2MP / UNO cannot detect camera

wangmasterX, 2019-01-02:
Hello, and Happy New Year!

I got an OV2640 Mini 2MP a while back and just plugged it into my UNO. I followed the instructions on the knowledge base, and opened ArduCAM_Host_V2. It tells me that the COM port is opened but no initialization of SPI or detection of the ArduCAM. See below:

[image]

My connections are correct, I've triple checked them. See below:

[image]
[image]

I'm using the ArduCAM_Mini_2MP_OV2640_functions file as instructed. Is there a way to check if the program is detecting my Arducam correctly? Not sure if i'm doing something wrong or my arducam is faulty.

wangmasterX, 2019-01-02:
Attached more pictures of my connections for clarity:

[image]
[image]

wangmasterX, 2019-01-02:
I ran the code via Serial Monitor and found that the SPI connection might be the issue, because I keep getting `ACK CMD SPI interface Error! END`

Is there an issue with my board? Because I'm quite sure my connections are correct.

wangmasterX, 2019-01-02:
I ran ArduCAM_Mini_2MP_OV2640_functions on the IDE and only got this:

```
ACK CMD ArduCAM Start! END
ACK CMD SPI interface OK. END
ACK CMD OV2640 detected. END
```

So I guess the SPI and camera are detected. But I'm still not getting any response from ArduCAM_Host_V2.

UCTRONICS, 2019-01-02:
@wangmasterX 
Are the serial port number and baud rate right?

wangmasterX, 2019-01-03:
Im using an Uno clone, but my crystal oscillator IS 16MHz, unless you're referring to another oscillator. I have attached an image to show you:

[image]

The program was able to serial print at 921600 baud rate, so i'm not sure baud rate is the issue here.

elvinosmanov, 2019-06-15:
Did you find the solution of this problem?

UCTRONICS, 2019-06-17:
Have you tried to set the baud rate to 115200？For some pc, the current of the usb port is low, which can't support high speed baud rate,please try to modify the baud rate ih the code.

UCTRONICS, 2019-06-17:
What's more, I advise you use short cable line, if the cable line is too long, it will effect the signal of image data.