ArduCAM/Arduino #385: Wemos D1 Mini Usage with OV2640 Mini 2MP

JoustingLion, 2019-01-12:
I have set up a Wemos D1 Minbi with an ArduCAM 2MP Mini module using the arduino IDE sketch and wiring diagram listed here: 

https://www.instructables.com/id/ArduCAM-Mini-ESP8266-Web-Camera/

I always get this error when I try to start the program and I am not able to receive any images from the webserver:
21:25:25.089 -> SPI1 interface Error!
21:25:25.089 -> Can't find OV2640 module! pid: 254
21:25:25.222 -> 
21:25:25.222 -> 
21:25:25.222 -> Share AP: arducam_esp8266
21:25:25.222 -> The password is: APpassword
21:25:25.222 -> 
21:25:25.222 -> AP ip
21:25:25.222 -> 192.168.4.1
21:25:25.222 -> Server started
Can someone please help me resolve this situation? I have tried to use this Arducam on a larger NodeMCU, but once I used the same code and wiring on this Wemos D1 Mini, it no longer functions.

UCTRONICS, 2019-01-13:
@JoustingLion 
Please ensure the hardware is correct. Checking the SPI interface and I2c interface

JoustingLion, 2019-01-13:
Sorry, i ended up having loose connections.

closed 2019-01-13 by JoustingLion