ArduCAM/Arduino #386: Multiple libraries were found for SD.h

molder26, 2019-01-13:
Hello, when i try to compile the example of Arducam_ESP8266_UNO_V1_Capture2SD i have an error of sd library

Arduino: 1.8.5 (Windows 7), Board: "ArduCAM ESP8266 UNO, 80 MHz, 921600, 4M (3M SPIFFS)"

WARNING: library Wire claims to run on (esp8266) architecture(s) and may be incompatible with your current board which runs on (ArduCAM_ESP8266_UNO) architecture(s).
WARNING: library SPI claims to run on (esp8266) architecture(s) and may be incompatible with your current board which runs on (ArduCAM_ESP8266_UNO) architecture(s).
In file included from C:\Users\Visual\Documents\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,

                 from C:\Users\Visual\Documents\Arduino\libraries\SD\src/utility/SdFat.h:29,

                 from C:\Users\Visual\Documents\Arduino\libraries\SD\src/SD.h:20,

                 from C:\Users\Visual\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.4\libraries\ArduCAM\examples\ESP8266\ArduCAM_ESP8266_UNO_V1_Capture2SD\ArduCAM_ESP8266_UNO_V1_Capture2SD.ino:17:

C:\Users\Visual\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.

 #error Architecture or board not supported.

  ^

Multiple libraries were found for "SD.h"
 Used: C:\Users\Visual\Documents\Arduino\libraries\SD
 Not used: C:\Users\Visual\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.4\libraries\SD
 Not used: C:\Program Files\Arduino\libraries\SD
exit status 1
Error compiling for board ArduCAM ESP8266 UNO.

How i can fix this error?
Regards

i0annis, 2019-09-10:
Hi, I had the same issue on a different program.
Deleting the folder C:\Users\...\Documents\Arduino\libraries\SD solved it for me.

bipinpcl22, 2019-11-10:
hi,
same issue 
but after deleting the folder when i recompile it again the files created in this folder and same error occured please sugess

bunglex, 2019-11-10:
Hi, 
Same for me too.
Its driving me mad, i have no idea why the issue came about..
ive uninstalled arduino, removed all temp files etc and as soon as i install the library i need i get the multiple library error

UCTRONICS, 2019-11-11:
@bunglex 
Have you try to delete all of the C:\Users...\Documents\Arduino folder?

bipinpcl22, 2019-11-11:
yes we deleted all C:\Users...\Documents\Arduino folder but as we compile the directory arduino automatic created but its blank no files are available in it . 

all libraries are in c program file

UCTRONICS, 2019-11-12:
@bipinpcl22 
 If so, I advise you delete the Arduino IDE and re-install it.

Tibertsmits, 2019-12-17:
Same problem here. I tried all options above, but nothing seems to work

ArducamTeam, 2019-12-18:
The best method is to completely remove the Arduino IDE related folders and libraries and then reinstall the Arduino IDE and the Arducam esp8266 uno board.

eneserd11e, 2020-12-13:
Same for me too but ı updated esp8266 configuration. ıt works.  ı don't take same error anymore

marcboon, 2020-12-15:
> Same for me too but ı updated esp8266 configuration. ıt works. ı don't take same error anymore

Can you please explain in more detail what you did?

nor2101, 2021-05-13:
Having the same problem with the latest 1.8.14 windows client and the esp32 package version 1.0.6. Arduino now selects the default SD library instead of the one in the esp32 package. Deleting the SD subfolder in the installation directory works as a temporary workaround, but it's not ideal.

kingofnull, 2021-06-24:
How do you install the lib try installing from repo and board manager.

rjsaurav13, 2023-03-21:
C:\Users\Visual\Documents\Arduino\libraries\SD 
delete the Arduino folder, it worked for me

Ivanhoe100, 2023-08-14:
I did a   full IDE install of 2.2.1 windows version after getting the same problem.  This included removing all registry lines that contained "Arduino" and any and all Arduino files and directories on my C: drive. I still ended up with the same error except this time it said:
  Used: C:\Users\XXXXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\SD
  Not used: C:\Users\XXXXXX\AppData\Local\Arduino15\libraries\SD
In my case I think the SD.h in the ESP8266 lib. and the standard lib. are different and the IDE had to figure that out.  At least this time it only loaded the right SD.h for my board.