ArduCAM/Arduino #387: Multi capture on arducam esp8266 UNO V2

molder26, 2019-01-14:
Hello, i modify the example Arducam Shield V2 Multi Capture2SD, but some times works well and some times i get an error:

ArduCAM Start!
SPI interface OK!
OV5642 detected.
SD Card detected.
start capture!
CAM Capture Done.
capture total_time used (in miliseconds):1007
The fifo length is :547464
OK
OK

Soft WDT reset

ctx: cont 
sp: 3ffefa00 end: 3ffefe40 offset: 01b0

>>>stack>>>
3ffefbb0:  000000ff 00003e58 0000012c 402038bb  
3ffefbc0:  3ffeea84 3ffeec48 000000c0 4020332c  
3ffefbd0:  00000010 3ffeea84 3ffeea88 3ffeeb00  
3ffefbe0:  3ffeeb08 3ffeea88 3ffeea84 40203d45  
3ffefbf0:  3ffeeb08 00000000 3ffeea88 40203db5  
3ffefc00:  3ffeeb04 3ffeeb08 00000001 40204785  
3ffefc10:  0005c3e7 00000100 3fff0564 3fff0564  
3ffefc20:  00000100 00000100 3ffefc90 4020467d  
3ffefc30:  3ffeeb02 00000000 00000000 3ffeea98  
3ffefc40:  00000000 00000100 00000001 0000003d  
3ffefc50:  00006c3d 00000000 00000000 00008000  
3ffefc60:  00000002 3ffeea98 00000000 0000004e  
3ffefc70:  000549e7 00000000 3ffefdb4 402034a6  
3ffefc80:  000549e7 3ffefde0 3ffeea0c 4020243c  
3ffefc90:  ff0d2932 d041c300 fe946e1f 04f947b9  
3ffefca0:  b5bf88d4 f0af1a7e dfdaae1a e57160c3  
3ffefcb0:  3fcdf7ee 42fed2ad 9e4a6439 3962f820  
3ffefcc0:  39b74f68 fc110fa7 3f231d2e a58c279d  
3ffefcd0:  32cb81fc c347f155 1f17eb05 2ede4a51  
3ffefce0:  fa0f62f8 a77afc9f 42fe90eb f1105979  
3ffefcf0:  e20fc377 d37a1c83 17fe8c8f 8f00ffa0  
3ffefd00:  d797e25c 3e00ff63 1922c9c5 bcf0773c  
3ffefd10:  3f420b1c 6f5a55e0 3c337c1b bcfe66c7  
3ffefd20:  bf1a6b1a 249cd8e7 f8373e50 6745bc70  
3ffefd30:  335f7f20 8c2faf8a 9c481e7e 1d5fc85a  
3ffefd40:  4a00ff4b 00ffadaf 31935c70 e04bfc6e  
3ffefd50:  4f163969 f5c21fbb 6f12fe5f 2df96e05  
3ffefd60:  b1ef3fee eb4fe3c7 ec47f94f ab7830e4  
3ffefd70:  b00df1c1 967a2ac5 553feb3b 7537fcd6  
3ffefd80:  7e6cc9b7 67fd691d ccc8a1fb 022d3589  
3ffefd90:  3ffe8460 00000000 000003e8 00000001  
3ffefda0:  706a2e33 00000067 00000000 37303000  
3ffefdb0:  3fff0564 3ffe8460 00000000 000003e8  
3ffefdc0:  00000001 706a2e33 00000067 00000000  
3ffefdd0:  3ffeed00 3fff0564 706a2e33 40200067  
3ffefde0:  00000000 00000010 00780301 00000000  
3ffefdf0:  00000002 00000010 3ffeede8 40204e7c  
3ffefe00:  3ffeede8 3ffeea10 3ffeede8 3ffeee0c  
3ffefe10:  3ffeede8 3ffeea10 3ffeea1c 402025e5  
3ffefe20:  3fffdad0 00000000 3ffeee04 4020515c  
3ffefe30:  feefeffe feefeffe 3ffeee20 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

The camera is connected directly to arducam esp8266 UNO V2.
I change the size of picture to check but i all time get this error.
Regards

Pablo

UCTRONICS, 2019-01-14:
@molder26 
Please add some yield() when you transfer image data.

molder26, 2019-01-14:
exactly where? i take the example sketch Arducam Shield V2 Multi Capture2SD.
Thank you in advance
Regards

Pablo

UCTRONICS, 2019-01-14:
@molder26 
[image]

molder26, 2019-01-14:
It's possible make more than 6 captures, i am needing also like 10 captures before save it.
How i can do it?
Regards

[demo]<https://user-images.githubusercontent.com/27711713/51114400-2b017480-1840-11e9-8cf9-3e0716edbadb.png>

molder26, 2019-01-14:
Another thing is when i capture 6 frames in buffer in max quality it take around 90 sec to save it to sd card, it's possible do it in less time?

molder26, 2019-01-15:
Hello, i have the last version, i think my problem is the time to save in the sd card, because if i take 6 frames in max quality (2592x1944) with JPG it take around 88 sec to save it.
Do you have any way to do it more fast?
Regards