ArduCAM/Arduino #390: Program stucks at "Start Capture"

ammarqs, 2019-01-22:
I have 2 of the OV2640 camera module. and have been trying to make them work for quite some now. I have tried loading all the program related to the Arducam_Mini......... and all have hang at some point or the other. I will set this discussion just to follow the video streaming program and the capture2SD program. 
Initially i thought i was my controller SPI that might have issues so i bought 2 more just to see if something is wrong, but same response. When i run the SD card program on its own without the camera, it runs fine, both read and write are fast. 
As soon as i introduce the camera, my program freezes at "Start capture" for a few minutes and then starts with a few click and freezes again and this keeps going on. Also, if it does take a picture successfully, it does not write to the sd card. AND for some reason if it does write something on the SD card, the file is corrupted.
So too much going on but nothing is right. Any suggestions.?

ammarqs, 2019-01-22:
SOLUTION:
ArduCAMs DO NOT work with 5 Volts power. Only with 3.3 Volts.
Even though the documentation says to connect it to 5 volt power supply, you will feel the IC heating up and no response when taking pictures. So connect it to 3.3 volts and it will work perfectly.

ArduCAM, 2019-01-23:
Can you show us the photos of your hardware setup?
Thanks.

ammarqs, 2019-01-23:
I also have a LTE modem connected to this test.
[image]
[image]
[image]
[image]