ArduCAM/Arduino #392: Live video stream

ammarqs, 2019-03-05:
Is it possible to do a live video stream over ethernet to a web browser or mobile app with arducam and arduino mega, along with an ethernet shield?
Obviously i dont have an mobile app yet, but we considering to develop one if we know for a fact that this would work?

UCTRONICS, 2019-03-06:
@ammarqs 
Arduino UNO board does not support ethernet. You can consider the esp8266 board.

ammarqs, 2019-03-06:
That is why i mentioned ethernet shield..

UCTRONICS, 2019-03-06:
@ammarqs 
Why not use Arduino Yun board or ESP8266 UNO board? I have not done any test on UNO using 
Ethernet. You should try it by yourself.

ammarqs, 2019-03-06:
@UCTRONICS 
I have ordered the ESP8266 NODEMCU. Should be receiving it tomorrow.
If you have tried this already, where do i start?
I am trying to hook it up with one of the arducams and do live streaming remotely.

UCTRONICS, 2019-03-07:
@ammarqs 
OK, You can refer to this link to get the detail operation
https://github.com/ArduCAM/ArduCAM_ESP8266_UNO