ArduCAM/Arduino #397: Tips wanted wrt image quality

jeroenhouben, 2019-03-28:
Hi,

I'd really appreciate some tips regarding image quality on the `OV564`. I'm using it with an ESP32 Arducam dev board.

These are my settings:

```
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_2048x1536);
  myCAM.OV5642_set_Compress_quality(high_quality);
```

The image this produces are just not very good. Blurry, bad contrast. Distorted.

[image]

Is this what can be expected? I'm looking for tips to improve img quality.

- Use a different host (feather M0 ?)
- Use a different lense?
- Change settings (all set to auto/default atm)

help much appreciated

ArduCAM, 2019-03-28:
The image is out of focus, you have to unlock the lens and fine tune the lens back and forth to get the best focus. The default lens FOV is a little bit wide, so the corner is  distroted, you can try our low distrotion lens if you need that,
http://www.arducam.com/new-arrival-arducams-new-low-distortion-lenses/

jeroenhouben, 2019-03-28:
Thanks. 

Which lens would you recommend for low light or flashlight photography of people/portraits?

ArduCAM, 2019-03-28:
We don't recommend to use this camera for low light condition, you need better illuminator.
For night vision you need the lens without IR filter, and should work with 850nm IR LED light.

closed 2019-03-28 by jeroenhouben