ArduCAM/Arduino #398: arducam 5MP OV5642

jopa95, 2019-03-29:
Hello
I'm using the code from the site ArduCAM_Mini_5MP_OV5642_Plus_Functions and the result in the serial monitor is :
[image]

I don't understand what are this these signs in the end and how to fix it.
Please help.
Thank you.

jopa95, 2019-03-29:
Someone help please.

ArduCAM, 2019-03-31:
All the strange characters are the JPEG data. You can't view the image through the serial monitor.
You have to save these JPEG data into a file. Please use our Windows GUI to view the image.

jopa95, 2019-03-31:
Thanks for answering.
How can I save the JPEG to a file?

UCTRONICS, 2019-04-01:
@jopa95 
You can use Host V2 application. or SD card. https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app

jopa95, 2019-04-01:
Is there any other way to save the JPEG?

UCTRONICS, 2019-04-01:
@jopa95 
You can store the image into your SD card. Referring to this demo 
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_Plus_Multi_Capture2SD/ArduCAM_Mini_5MP_Plus_Multi_Capture2SD.ino

jopa95, 2019-04-01:
I undestand, but I dont have a SD card, is there any way to store the JPEG on the computer directly?

VigorZero, 2019-07-17:
Same question.

UCTRONICS, 2019-07-18:
@jopa95  @VigorZero 
Yes, out Host V2 application supports store image into the local path.
[image]