ArduCAM/Arduino #400: MT9J001 Arducam compatible with Arduino (with or without USB2 camera shield)

MarcelRoehrl, 2019-04-02:
Hey,
we want to use the MT9J001 cams on our nrf52 Arduino boards. 
Would that be possible? Because the memorysaver.h file in the Arduino/ArdCAM library is not listing this camera, not even with the use of the v2 camera shield.
The description of the USB2 camera shield says that it supports this camera, but as far as I can tell it's not specified if it would work with an Arduino.
The camera shield also supports ROS, so would it be possible to run ROS on the Arduino to use the camera?
Further more, would it be technically possible to connect this camera directly to an Arduino board? If so, we would invest some time to try it.
Thanks in advance

ArduCAM, 2019-04-02:
MT9J001 is raw sensor, the image size is too big for Arduino to handle.
It should work with USB camera shield together and 
We suggest you to use Raspberry pi to work with this camera.

MarcelRoehrl, 2019-04-03:
Thanks for the fast reply, we've to use Arduinos because low power consumption is important for our project. We'll have to try the shields then