ArduCAM/Arduino #405: Need help in initializing Arducam

Kyanaam, 2019-04-13:
Hi,
I have set up Arducam Shield Mini 5MP Plus with Beaglebone Black Rev C. I am able to read/write Arducam registers through SPI and I2C. I have gone through OV5642 datasheet and SW application notes.

I am unable to get frame size anything different from 8 when I read registers 0x42~0x44. When I perform single read 10 times to read frame buffer I get all 0's for first 8 reads and then possibly some garbage value for remaining  reads.

I suspect wrong initialization of arducam. Now my problem is that I am a newbie and there are too many registers involved in initialization. I am including "ov5642_regs.h" in my source code. What are the registers that I should focus on? Any leads/clues would be really helpful.

Thanks in advance.

ArduCAM, 2019-04-14:
Please have a look at the following code, although it is quite old and prreliminary.
http://arducam.com/downloads/porting/BBB.rar 
Why you get 8 bytes frame size, it probably you forgot the set the inverse the VSYNC polarity register.

Kyanaam, 2019-04-14:
Many many thanks. I am able to get a good frame size and could see end of frame with 0xFF and 0xD9 as well. Setting VSYNC polarity register to active low was the key.

Kyanaam, 2019-04-27:
Hi,
I get a valid JPEG image from arducam on beaglebone black with above suggestion. I transfer this image to Ubuntu PC to view it but I see an image that is just black and no image. I have seen this or similar issue reported on many forum but no resolution. Is there no solution? I have tried including ov5642_regs.h from many places and with various init options. Is there any help available? Should I contact OmniVision?

Thanks.

ArduCAM, 2019-04-27:
Please upload the image and we can check what happened. OmniVision won't help.

Kyanaam, 2019-04-27:
[image]
Please find image attached.

Kyanaam, 2019-05-01:
Were you able to look into it? Any updates?

ArduCAM, 2019-05-01:
The image is good but only too dark I guess, once you initialize the camera you have to wait for few seconds for AWB, AE to be stable then start a capture.

Kyanaam, 2019-05-01:
When I introduce a delay of two seconds just before "start capture", I see a strange behavior- After capture is done I get very big Frame size, almost 4 times bigger. And when I read FIFO (one at a time and not a burst read) I never get FFD9 which should be there at the end of every JPEG image. This is not the case without any delay.

Kyanaam, 2019-06-15:
It works now. I added 3 seconds of delay after initializing the camera.

closed 2019-06-15 by Kyanaam