ArduCAM/Arduino #410: Ov5642 5mp cam

jopa95, 2019-04-29:
I use the ov5642 5MP  plus cam and your example from the site of the o5642 plus functions.
If I start your example every time stops on the same place, it never goes to “capture done”
I include all necessary libraries.
Someone help
Here is also a picture of serial monitor.
[image]

jopa95, 2019-04-29:
Help someone?

UCTRONICS, 2019-04-30:
@jopa95 
Which demo you are using? Our Video Streaming demo should use the Host V2 application .

jopa95, 2019-05-03:
It take sometimes a photo , but usually it stops by cam start single shot . I think the profram don’t get into the if myCamgetbit function , dont know why? Can someone help

closed 2019-05-04 by jopa95