ArduCAM/Arduino #412: Shield V2 and Arduino Uno - No camera playback

JulenWF, 2019-05-02:
I am using the Arduino UNO and the ArduCAM Shield V2, with the OV2640 Camera.

I've followed tutorials that should lead me to showing an image on the Shield V2, everything seems to go fine, the feed onto the screen goes black like it shows in the video however it stays that way and does not give me camera feed.

I followed https://www.youtube.com/watch?v=_Dk5vYtqT5M&t=209s

This is what came out of it.

[image]

[image]

"In file included from C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Shield_V2\ArduCAM_Shield_V2_Camera_Playback\ArduCAM_Shield_V2_Camera_Playback.ino:18:0:

C:\Program Files (x86)\Arduino\libraries\ArduCAM/ArduCAM.h:114:0: warning: "pulse_high" redefined

 #define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);

 ^

In file included from C:\Program Files (x86)\Arduino\libraries\UTFT4ArduCAM_SPI/UTFT_SPI.h:189:0,

                 from C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Shield_V2\ArduCAM_Shield_V2_Camera_Playback\ArduCAM_Shield_V2_Camera_Playback.ino:15:

C:\Program Files (x86)\Arduino\libraries\UTFT4ArduCAM_SPI/HW_AVR_SPI_defines.h:5:0: note: this is the location of the previous definition

 #define pulse_high(reg, bitmask) sbi(reg, bitmask); UTFT_cbi(reg, bitmask);

 ^

In file included from C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Shield_V2\ArduCAM_Shield_V2_Camera_Playback\ArduCAM_Shield_V2_Camera_Playback.ino:18:0:

C:\Program Files (x86)\Arduino\libraries\ArduCAM/ArduCAM.h:115:0: warning: "pulse_low" redefined

 #define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);

 ^

In file included from C:\Program Files (x86)\Arduino\libraries\UTFT4ArduCAM_SPI/UTFT_SPI.h:189:0,

                 from C:\Program Files (x86)\Arduino\libraries\ArduCAM\examples\Shield_V2\ArduCAM_Shield_V2_Camera_Playback\ArduCAM_Shield_V2_Camera_Playback.ino:15:

C:\Program Files (x86)\Arduino\libraries\UTFT4ArduCAM_SPI/HW_AVR_SPI_defines.h:6:0: note: this is the location of the previous definition

 #define pulse_low(reg, bitmask) UTFT_cbi(reg, bitmask); sbi(reg, bitmask);

 ^

Sketch uses 21928 bytes (67%) of program storage space. Maximum is 32256 bytes.
Global variables use 1135 bytes (55%) of dynamic memory, leaving 913 bytes for local variables. Maximum is 2048 bytes."

I hope you can help my case!

JulenWF, 2019-05-02:
I've solved it now, all that was missing was an SD Card, without it I was not able to get video output.

closed 2019-05-02 by JulenWF

UCTRONICS, 2019-05-02:
@Great to hear you have solved your problems.
Notice, you just can only enable one sensor.
[image]