ArduCAM/Arduino #413: Image transmission using Bluetooth on Arduino and Android

ghks4544, 2019-05-03:
I think i connected android and OV5642 with Bluetooth.
Serial port take to me High quality jpeg image.
But in bluetooth communication, i think wrong baudrate is setted and not synchronized with android.
And bluetooth communication take to me garbage and noised jpeg image.

If you see my question, and answer me. I'll provide my source code. 
And I also i am going to finding my foolish fault .

Any help is appreciated!

UCTRONICS, 2019-05-05:
@ghks4544 
Using the blue tooth, I advise the baud rate is less than 115200.

ghks4544, 2019-05-15:
@UCTRONICS 
Thanks!

As you advised, Baudrate was the problem.

I will try not to make these minor mistakes.

closed 2019-05-15 by ghks4544