ArduCAM/Arduino #415: Arducam Mini 5MP Plus firmware

giogrimo, 2019-05-20:
Hi, 
I recently bought the Mini 5MP Plus and I found a difference in firmware version. In particular, as shown in the picture (taken from ArduCAM_Mini_5MP_Plus_OV5642_Camera_Module_Hardware_Application_Note.pdf), the version should be 0x62 (read from address 0x40) but I read 0x73. 
What is the difference in the new firmware? It exist and updated document?

Thanks for the help :)

[image]

ArduCAM, 2019-05-22:
The camera firmware version is updated from time to time. We will update the document very soon:
http://www.arducam.com/docs/spi-cameras-for-arduino/hardware/arducam-chip/

giogrimo, 2019-05-22:
ok perfect. 
Which firmware version is referring the documentation in the link?