ArduCAM/Arduino #420: Arducam mini 2MP OV2640 does not start (functions demo + Host)

NicoletaGabriela, 2019-06-27:
Hello! 

Before I start I just want to say that I did try to find a solution for my problem, I just could not find one. 
I have an Arducam mini 2MP OV2640 I just want to try it out with with the example sketch ArduCAM_Mini_2MP_OV2640_functions. I did change memorysaver.h , The CS pin I use is 10 so I changed that too in the code (I tried pin 7 too but it does not work either), I do not see a problem regarding the connections. All i get in the serial monitor is just garbage and when I try the host app The COM does open but the Arducam does not start. I will attach some pictures. 
In case someone has an idea what the problem might be  I would appreciate a response. 

Thank you! 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ArducamTeam, 2019-06-28:
I see that your serial port has content output, may be the baud rate is incorrect. You can check the baud rate in the demo, make sure the baud rate is consistent with the serial monitor.

ArducamTeam, 2019-06-28:
Hi!

Your serial port has garbled output, so it may be the problem of serial port. In the example sketch ArduCAM_Mini_2MP_OV2640_functions, the baud rate is 921600.I suspect that your board may not support such a high baud rate, you can try to change the baud rate to 115200, and then try to open the serial port again.If there is still no output, please responed again.

NicoletaGabriela, 2019-06-28:
Unfortunately it does not work with any other baud rate.

ArducamTeam, 2019-06-28:
You might write a simple serial port test example first, without adding anything else, and check if the serial port can be output normally.

NicoletaGabriela, 2019-06-28:
The serial port works for 115200 bps, but I noticed that I can not set the baud rate 921600 in the serial monitor. 
[image]
[image]

ArduCAM, 2019-06-28:
@NicoletaGabriela 
It seems that when the camera is attached, your board won't work properly. Please try to connect the camera to your board and test the serial function using your test code.
If it doesn't work, maybe your hardware design exist problems. What's the type of your power supply chip? You can attach us your schematic diagram and our engineer will help you check it in detail.

NicoletaGabriela, 2019-06-28:
Even when the camera is attached, the simple serial test code is still working.

ArduCAM, 2019-06-28:
@NicoletaGabriela 
Please try this demo to test if the camera can be detected. 
The CS is 7
[ArduCAM_Mini_Camera_Test.zip](https://github.com/ArduCAM/Arduino/files/3338548/ArduCAM_Mini_Camera_Test.zip)

NicoletaGabriela, 2019-06-28:
The camera was detected with that test program and after that I just set the serial port rate in the ArduCAM_Mini_2MP_OV2640_functions to 115200 without the #if block and then it worked. 
Thank you very much for the help! 
[image]
[image]
[image]

ArduCAM, 2019-06-28:
@NicoletaGabriela 
Great to hear your progress.