ArduCAM/Arduino #421: Arducam 2MP plus with SD card, spi error

Donovator122, 2019-07-02:
Hello, I am working with an arduino uno, the [arducam OV2640 mini 2MP plus](https://www.amazon.com/gp/product/B012UXNDOY/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1), and a[ micro SD module](https://www.amazon.com/Interface-Arduino-Raspberry-Breakout-Cable%EF%BC%88Pack/dp/B07MTTLF75/ref=sr_1_7?keywords=micro+sd+card+arduino&qid=1562035529&s=gateway&sr=8-7). I have tested just the camera using the host app and it works perfectly. However, when I connect both the camera and sd card, I receive a "spi interface error". Once I disconnect power to the SD card, it recognizes the camera. Past this point, the fifo size is oversize when the SD card is connected. The code I am using is the [mini-sd code](https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_2MP_Plus_Multi_Capture2SD/ArduCAM_Mini_2MP_Plus_Multi_Capture2SD.ino). 
I am not very experienced with SPI and am not sure if I need a line code to turn each on or off?
The wiring diagram is attached.
Thank you for any help!
[minicam:sd.fzz.zip](https://github.com/ArduCAM/Arduino/files/3348092/minicam.sd.fzz.zip)

ArducamTeam, 2019-07-02:
Hello,it should be your SD card slot design problem, does not support the three-state function, you can change a card slot or redesign the hardware circuit of SD card.

Donovator122, 2019-07-04:
What would be the best way to go about doing that?

ArducamTeam, 2019-07-04:
I recommend that you use our sd slot.

Donovator122, 2019-07-04:
I will try that, is there any way to reprogram what I have currently?

UCTRONICS, 2019-07-05:
@Donovator122 
Due to the poor design of some SD card slots, the SPI bus is not released, which cause the bus can't connect multi device. So I advise you change another SD card slot.

Donovator122, 2019-07-06:
Great, thank you

closed 2019-07-06 by Donovator122