ArduCAM/Arduino #422: Arducam 5MP OV5642 issue for raw images over 1280x960

MarcelRoehrl, 2019-07-05:
Hey,
I try to get full sized raw images from this camera, but I get strange data for images over 1280x960 resolution. Before that I can decode them just fine, but for higher resolutions the images get mirrored at the X axis (height axis) and the color decoding also has some errors.
I decoded the image from bayer to rgb using all variants of this document (page 99)
https://www.uctronics.com/download/cam_module/OV5642DS.pdf

0x0: BGBG... / GRGR... works perfect for pictures in small resolutions.
I tried every other variant (0x0 - 0x3) for the higher resolutions, but I still get bad images.
I also tried to mirror the image (correct it) before decoding it and fiddled with the encoding a bit but I still dont get good images.

Here are some of the images I get from the sensor (all converted from .bmp to .png to display them here)

Here is a perfectly fine raw image (censored) at 1280x960:
[image]

Following are all at 1920x1080 (same problem with 2560x1440)
not mirrored with 0x0 encoding:
[image]

Mirrored before encoding with 0x0:
[image]

Mirrored before encoding with 0x1 (same color results as not mirrored with 0x0):
[image]

X (height) and Y (width) axis mirrored before encoding with 0x1 (because the camera was upside down while taking every image, even all the low resolution ones):
[image]

I got the best result with coming up with my own bayer encoding (GBGB.../GRGR...) with the unmirrored version of the image, but can adjust that for the mirrored one:
[image]

Last one looks good when displaying it on the actual size on your monitor, but if you zoom in or out you can see the errors.

MarcelRoehrl, 2019-07-05:
Because it get lost a litte bit here is an image of the zoomed in .bmp of the last image 
[image]

UCTRONICS, 2019-07-08:
@MarcelRoehrl 
I have tested it and the configure does not every well, I will try to optimize the sensor configuration.

MarcelRoehrl, 2019-07-11:
Thanks, colors are not that important. A black and white image would be fine too. As you can see the higher resolution images are a lot darker than the 1280x960 image (because of the wrong bayer interpretation), a lot of details get lost in the dark areas.

UCTRONICS, 2019-07-12:
@MarcelRoehrl 
I have updated the config file, please try our new sensor config.

MarcelRoehrl, 2019-07-15:
Thanks, it got a lot better now. Image is still mirrored but that's no problem. 
The colors are still not accurate (still a bit greenish but less than before), but that's ok.
It's still a little bit too dark and some information get lost in the dark areas (even after cranking up the the brightness and contrast afterwards, which also reduces the greenish touch. Do you think you can optimize that as well?
Here is an new 2592x1944 image (converted to png) with not enough details in the dark:
[image]

Same picture with cranked brightness and contrast:
[image]

Here an image with better details in the dark areas:
[image]

Same with cranked brightness and contrast:
[image]

I cant adjust the colors to let them look more natural, so there seems to be something off.

UCTRONICS, 2019-07-16:
@MarcelRoehrl 
Strange, I test the config file on our platform and It seems normally. It is auto exposure mode, so you can't adjust the brightness, You can set it to manual exposure mode and set the brightness.
[image]

MarcelRoehrl, 2019-07-17:
I've to test it in the real environment too see if I get enough brightness with the auto exposure and will try manual exposure otherwise, thanks.
Your screenshot shows that you took the picture with a resolution of 1280x960, for that res my colors too look natural. Do you get the same colors with a higher resolution (2592x1944)?