ArduCAM/Arduino #423: ArduCAM_Host_V2.exe + BMP

bakerbrown, 2019-07-08:
When selecting the SetToBMP you get a very psychedelic output that is flipped vertically.

bakerbrown, 2019-07-08:
[files.zip](https://github.com/ArduCAM/Arduino/files/3367974/files.zip)

UCTRONICS, 2019-07-08:
@bakerbrown 
Do you use the newest source code and  host v2?

bakerbrown, 2019-07-09:
@UCTRONICS Yes, am I reporting the same issue? https://www.arducam.com/rgb565-format-issues/

UCTRONICS, 2019-07-09:
@bakerbrown 
Maybe, are you using https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_2MP_Plus_VideoStreaming/ArduCAM_Mini_2MP_Plus_VideoStreaming.ino demo ?

bakerbrown, 2019-07-09:
Have tried:
ArduCAM_Mini_2MP_OV2640_functions
ArduCAM_Mini_2MP_Plus_functions
both output the same corrupt BMP images! Think it more likely that the error is within ArduCAM_Host_V2.exe Where can I find the source to that application?

UCTRONICS, 2019-07-09:
@bakerbrown 
We don't provide the application source code. Please attach me  your hardware setup and I will help your camera version. If you are using the old version, you should delete the first data when reading the image data.

bakerbrown, 2019-07-09:
@UCTRONICS Thanks for your quick responses. "We don't provide the application source code." Sorry the licence is missing, so I assumed it was released under Qt's GPL, LGPL not the Commercial license model?

The camera is Mini-2MP-Plus rev A

bakerbrown, 2019-07-09:
After more experimentation you can almost recover the image by inserting a NULL after the header. I would strongly advise that the error is in the application that saves the .BMP file and not in the driver.

UCTRONICS, 2019-07-09:
@bakerbrown 
Thanks for your advise, for our newest version, it does not need to add or delete a byte data. we have optimized our hardware.

bakerbrown, 2019-07-10:
@UCTRONICS which "newest version"? I have only just bought the hardware and am looking to purchase another 20+ units for the current project, this will not happen if I am unable to resolve this issue.

Please advise which hardware version should I be purchasing, where should I purchase it from?

UCTRONICS, 2019-07-10:
@bakerbrown 
The version you are using is the newest verson Mini 2MP Plus Rev A. Tomorrow, our engineer will test it again and reply you as soon as possible.

ArducamTeam, 2019-07-11:
@bakerbrown 
Hello,we tested two examples with Arduino UNO and Mini 2MP PLUS.
Test examples: ArduCAM_Mini_2MP_Plus_VideoStreaming\ArduCAM_Mini_2MP_Plus_functions.
The two examples are perfect for both jpeg shooting on host v2 and BMP viewing, except the BMP image is rotated 180°.Below are two screenshots of my test.
[image]
[image]