ArduCAM/Arduino #427: Unable to record raw video at higher resolution

Kyanaam, 2019-07-11:
Hi,
I have set up Arducam Shield Mini 5MP Plus with Beaglebone Black Rev C. 
I can record a very short video clip (RAW YUV422 format) and play it on my desktop computer. But the problem is whatever settings I use from "ov5642_regs.h" I never get a video at higher resolution than 320x240. Is there a way to achieve it? "OV5642_1080P_Video_setting" gives multiple JPEG frames which I can't play at all.

Any help would be appreciated. Thanks in advance.

UCTRONICS, 2019-07-12:
@Kyanaam 
Hi,
If you want to get higher resolution, firstly, you should use  myCAM.OV5640_set_JPEG_size(); function to set the resolution.
secondly, you should change the size. the default is 640x480  (0x80, 0x02, 0x00, 0x00, 0xe0, 0x01)
[image]
Let me know if you need more help

Kyanaam, 2019-07-12:
It doesn't work still. Let me put it this way: 
When I initialize for qvga preview using below configuration, I get a good 320x240 video clip in YUV422 format which I can display on my computer monitor.
wrSensorReg16_8(0x3008, 0x80);
wrSensorRegs16_8(OV5642_QVGA_Preview); 

But no other configuration seems to work. I used configuration from  myCAM.OV5640_set_JPEG_size() for 640x480 as below but what I played that all I could see was some Rainbow color in a 640x480 window.
//wrSensorRegs16_8(OV5642_1280x960_RAW);
//wrSensorRegs16_8(OV5642_640x480_RAW);
//wrSensorRegs16_8(ov5642_640x480);

Please let me know if I am missing something here.

UCTRONICS, 2019-07-12:
@Kyanaam 
It does not support raw video. It just encoder it to mjpeg video