ArduCAM/Arduino #430: ESP8266 + Arduino Uno + OV5642

Seaatm, 2019-07-19:
Hello, I want to take a photo and save it into a web server or something like that, the fact is that I want to be able to access the photo taken later on. This is my setup:
-Arduino UNO
-OV5642
-ESP8266
[image]

I want to use the "ArduCAM_ESP8266_OV5642_Capture" example but I am not really sure if this is going to work because the example works with the ArduCAM ESP8266 UNO board,

ArducamTeam, 2019-07-20:
@Seaatm 
Hi, I advise you to use our ArduCAM_ESP8266_UNO development board with the "ArduCAM_ESP8266_UNO_Capture" example and you will realize your functionality easily.

ArducamTeam, 2019-07-20:
@Seaatm 
Hi, Your ESP8266 module was flashed AT firmware,it can not flash our example code directly.Therefore,I advise you to use our ArduCAM_ESP8266_UNO development board with the "ArduCAM_ESP8266_UNO_Capture" example and you will realize your functionality easily.

UCTRONICS, 2019-07-20:
@Seaatm 
Hello, If you just want to be able to access the photo taken later on, you just need to use UNO with our Host V2 GUI, which supports preview and store image to local path. Visit https://github.com/ArduCAM/Arduino to get detail operation. Just use our videostreaming demo.
We don't have demo for the 8266 module with AT firmware . Our ArduCAM_ESP8266_OV5642_Capture demo is used for esp8266 with noduMCU firmware. So if you want to use this demo, I advise you to use our ESP8266 UNO  board which can replace your UNO and 8266 module.