ArduCAM/Arduino #431: Error compiling for board Nucleo-144

JakobTernes, 2019-07-22:
Hello everyone,
I'm currently working on taking a picture with the nucleo 144 STM32 H743ZI board and an OV5642 5MP camera module by using the arduino IDE.
I have downloaded the required software in the boards manager and chose the exact board.
I have also downloaded the ArduCam library from here and edited the memorysaver.h file to my needs.
But when I try running the example program from the examples file in the ArduCam library, it doesen't compile it and puts out an error as you can see in image 1.
Image 2 contains the tools settings i used.
Btw: a simple blink program worked on the board, so I suspect there is something wrong with the compatibility of the board & library, but I'm really not sure there.

Thank you for your answers

Jakob Ternes

[image]
[image]

UCTRONICS, 2019-07-23:
@JakobTernes 
At present, the Arducam library  doesn't compatible with the stm32 platform.

JakobTernes, 2019-07-23:
Thanks a lot, that saved me quite some unnecessary work.