ArduCAM/Arduino #432: ArduCam Mini ov2640 BMP colors & orientation

maminej, 2019-07-25:
Hello, 

We are currently testing the Arducam mini with the 2MP OV2640. We are using the ArduCAM_Mini_2MP_OV2640_functions.ino on an arduino uno. We are getting good results when the camera is setup in JPG mode. However, when in BMP mode, the colors are not correctly set as well as the orientation. s
is this a known issue ? we are certain we have the latest version of code. can you please investigate 
[image]
[image]

UCTRONICS, 2019-07-26:
@maminej 
This is due to the pixel data bit error, for our new version camera, you don't need to lost the first byte.
Just go to the line 495  and comment the SPI.transfer()
[image]

maminej, 2019-07-29:
That fixed the problem . thanks.
Can you please explain what does the 0x00 SPI command do ?

UCTRONICS, 2019-07-30:
@maminej 
You are welcome. the 0x00 is dummy data which is in order to read data. You can get it from SPI protocol.